zephyrproject-rtos/zephyr #79533: Multiple BT Mesh tests are failing in main

aescolar, 2024-10-08:
**Describe the bug**
Multiple BT Mesh tests are failing in main

**To Reproduce**
For ex.
1. tests/bsim/compile.sh
2. tests/bsim/bluetooth/mesh/tests_scripts/dfu/dfu_cli_all_targets_lost_on_caps_get.sh
4. See error

**Expected behavior**
No failing tests

**Impact**
Blocked CI in PRs

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/11220193326/job/31201648682?pr=79262

**Environment (please complete the following information):**

 - OS: Ubuntu 22.04 (CI)
 - Toolchain (e.g Zephyr SDK, ...): Host gcc 11/12
 - Commit SHA or Version used: main as of now: 318b49570a77ce631dc3d97ad7e7a05df5459d8e

**Additional context**

aescolar, 2024-10-08:
CC @PavelVPV @alxelax

PavelVPV, 2024-10-08:
I will have a look

alxelax, 2024-10-08:
Just took a quick look, it seems import of keys is broken. I suspect some PSA related changes were committed without running mesh tests.

PavelVPV, 2024-10-08:
This commit broke mesh: https://github.com/zephyrproject-rtos/zephyr/commit/4b479016a792a24704352566bdbf726493dd08ae

alxelax, 2024-10-08:
> This commit broke mesh: [4b47901](https://github.com/zephyrproject-rtos/zephyr/commit/4b479016a792a24704352566bdbf726493dd08ae)

I wonder if this adds PSA ITS for posix platforms, or it still requires using files.

aescolar, 2024-10-08:
CC @tomi-font

PavelVPV, 2024-10-08:
Seems adding `select SECURE_STORAGE` to `config BT_MESH_USES_MBEDTLS_PSA` made it working. Checked on one test, checking other mesh tests now.

alxelax, 2024-10-08:
> Seems adding `select SECURE_STORAGE` to `config BT_MESH_USES_MBEDTLS_PSA` made it working. Checked on one test, checking other mesh tests now.

I corrected a bit my comment above. Do we need my workaround at all after this commit?

tomi-font, 2024-10-08:
Hey. The secure storage subsystem supports native platforms. I have tested with `native_sim`, but I think it has run in CI on `native_posix` as well.
I encourage you to switch to using it, and remove your custom implementation.

PavelVPV, 2024-10-08:
@tomi-font , it forces `SETTINGS` option: https://github.com/zephyrproject-rtos/zephyr/blob/318b49570a77ce631dc3d97ad7e7a05df5459d8e/subsys/secure_storage/Kconfig.its_store#L15.

How can I use Secure Storage without Settings?

PavelVPV, 2024-10-08:
@tomi-font , we need some kind of this: https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/subsys/secure_storage/psa/its/src/custom_store.c as one of Secure Storage ITS module implementation (call it `SECURE_STORAGE_ITS_STORE_IMPLEMENTATION_RAM`).

tomi-font, 2024-10-08:
> @tomi-font , it forces `SETTINGS` option:
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/318b49570a77ce631dc3d97ad7e7a05df5459d8e/subsys/secure_storage/Kconfig.its_store#L15
> 
> How can I use Secure Storage without Settings?

You cannot do with settings enabled?

> @tomi-font , we need some kind of this: https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/subsys/secure_storage/psa/its/src/custom_store.c as one of Secure Storage ITS module implementation (call it `SECURE_STORAGE_ITS_STORE_IMPLEMENTATION_RAM`).

Yeah, not using settings would require providing an alternate ITS store module implementation. But why can't you do with an actual storage solution in the first place?

PavelVPV, 2024-10-08:
I want to keep an option to build mesh without settings. Which prevents me adding `select/depends on SECURE_STORAGE` to `config BT_MESH_USES_MBEDTLS_PSA`. Neither I get an error or warning that running a firmware without `SECURE_STORAGE` but with `SETTINGS` enabled won't work because ITS implementation is missing. I have to add an assertion into mesh code to enable `SECURE_STORAGE` when `SETTINGS` is enabled by user which is not super convenient.

tomi-font, 2024-10-08:
It seems that your ITS implementation is only to run some tests. I don't exactly get why you'd want an option to build without settings.
But if this is breaking CI in PRs, maybe at least for now the best is to adjust what was done to [config-tls-generic.h](https://github.com/zephyrproject-rtos/zephyr/commit/4b479016a792a24704352566bdbf726493dd08ae#diff-35af03c2e234822e54b1a5ef8918c465ca47f246d1dc732740e125e6a2935732) to revert the behavior change for Bsim.
Also, FYI, in the (hopefully near) future an alternative ITS store implementation than settings will be provided, some time after [ZMS](https://github.com/zephyrproject-rtos/zephyr/pull/77930) has made it into Zephyr.

omkar3141, 2024-10-08:
> It seems that your ITS implementation is only to run some tests. I don't exactly get why you'd want an option to build without settings.

For testing purposes, we do build and run mesh on devices without Settings SS. Having settings enabled is sure way to kill the flash while performing some large scale tests with hundreds of thousands of messages getting processed on a node and each message potentially triggering a write operation. It is also possible to build a fully functional mesh application that pre-provisions itself and becomes functional at power-on - this does not Settings SS. If the end-product/system is designed in such a way that it never encounters power-cycle, it never needs Settings SS.

aescolar, 2024-10-08:
>> It seems that your ITS implementation is only to run some tests. I don't exactly get why you'd want an option to build without settings.

> For testing purposes, we do build and run mesh on devices without Settings SS. Having settings enabled is sure way to kill the flash while performing some large scale tests with hundreds of thousands of messages getting processed on a node and each message potentially triggering a write operation.

Have you considered using the flash simulator, with a small flash partition simulated in RAM?

PavelVPV, 2024-10-08:
> Have you considered using the flash simulator, with a small flash partition simulated in RAM?

IIRC there was some discussion about it but it has never been prioritized.

PR #79543 (2024-10-08): modules: mbedtls: Enable back Mbed TLS PSA Crypto Storage C
This partially reverts https://github.com/zephyrproject-rtos/zephyr/commit/4b479016a792a24704352566bdbf726493dd08ae to fix Bluetooth mesh tests with bsim.

Fixes #79533
commit 86b66654 (2024-10-08): modules: mbedtls: Enable back Mbed TLS PSA Crypto Storage C

This partially reverts https://github.com/zephyrproject-rtos/zephyr/commit/4b479016a792a24704352566bdbf726493dd08ae
to fix Bluetooth mesh tests with bsim.

Fixes #79533
closed 2024-10-08 by aescolar