zephyrproject-rtos/zephyr #79550: Wrong Assertion in sensor.h

mirkobeckerPxC, 2024-10-08:
This `__ASSERT(cqe->userdata != buf,
		 "consumed non-matching completion for sensor read into buffer %p\n", buf);` should be changed to `__ASSERT(cqe->userdata == buf,
		 "consumed non-matching completion for sensor read into buffer %p\n", buf);`

https://github.com/zephyrproject-rtos/zephyr/blob/41e4b5323ac3888c45763ed7574c892cf3907fb5/include/zephyr/drivers/sensor.h#L1067

henrikbrixandersen, 2024-10-08:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

fabiobaltieri, 2024-10-08:
Hey @mirkobeckerPxC, could you open a pull request with the fix? The contribution guideline is https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines. Thanks!

PR #79602 (2024-10-17): sensor: fix assertion
In sensor_read()
With rtio_sqe_prep_read() buf is assigned to userdata. The __ASSERT should check if the same userdata arrived with rtio_cqe_consume_block(). The assert condition was wrong.

fixes #79550
closed 2024-10-17 by nashif