zephyrproject-rtos/zephyr #79564: wifi: Use TLS credentials libraryy for 802.1x certificates

krish2718, 2024-10-08:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, 802.1x certificates for Wi-Fi Enterprise have to be passed during build time which is okay for developement, but for production we need a trusted storage based solution.

**Describe the solution you'd like**

Use the existing library https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/net/lib/tls_credentials to also handle 802.1x certifications. This library provides various backends RAM, PS etc.

**Describe alternatives you've considered**

None.

**Additional context**

jukkar, 2024-10-10:
See also #78048.

krish2718, 2024-10-10:
Thanks. The Wi-Fi credentials still doesn't support Enterprise (certs) and might end up using TLS credentials as a backend.

closed 2025-04-17 by krish2718