zephyrproject-rtos/zephyr #79569: Tests of various nxp boards fail after #78429

bdkrae, 2024-10-08:
**Describe the bug**
Tests of various boards featuring NXP soc´s fail after #78429 was merged. I tested mm_feather, teensy41, mimxrt1060_evk, frdm_mcxc242 but there should be even more.
Inserting `CONFIG_PLATFORM_SPECIFIC_INIT=y` in the boards defconfig-file fixes it, but seems to be a deprecated option.

**To Reproduce**
`./scripts/twister -c -b -p mm_feather -T samples/subsys/shell/shell_module`
delivers at least 1 error.

**Expected behavior**
Tests must succeed.

**Impact**
PRs cannot be merged due to failing tests, e.g. #78417

mmahadevan108, 2024-10-08:
Did you `west update` to get the latest NXP HAL

pdgendt, 2024-10-08:
I think this has been fixed on main already: #79558

bdkrae, 2024-10-08:
> I think this has been fixed on main already: #79558

Thanks, #79558 solves the issue

closed 2024-10-08 by bdkrae