zephyrproject-rtos/zephyr #79574: `counter_start()` has undefined behaviour

ajf58, 2024-10-08:
`counter_start()` has undefined behaviour, which this test relies on. It assumes:

1. `counter_start()` starts at zero.
2. Calling `counter_start()` will restart a running counter to zero.

The documentation of the API doesn't define this one way or another, but for some drivers (e.g. drivers/counter/counter_rpi_pico_timer.c:counter_rpi_pico_timer_start) this isn't the case, and running `test_valid_function_without_alarm` against that driver will failure. I suspect other drivers will similarly fail.

I think at least the documentation of the API needs to define the expected behaviour, and driver(s) will need to be updated to match.

_Originally posted by @ajf58 in https://github.com/zephyrproject-rtos/zephyr/issues/76527#issuecomment-2400618165_

ajf58, 2024-10-09:
Mentioning the counter maintainer @nordic-krch .

@nordic-krch , what's the intended behaviour of the counter interface when `counter_start()` is called? Do you agree that current implementations are inconsistent, i.e. some reset (restart) to zero, others will allow a running counter to continue to run. 

cc: @butok , @nashif , author and merger of the associated PR https://github.com/zephyrproject-rtos/zephyr/pull/76527 respectively.

nordic-krch, 2024-10-09:
Yes, it is currently unclear. We probably have few options to chose.
One thing that comes to my mind is to add `counter_pause()` which stops the counter but does not reset it (and `-ENOTSUP` is returned if it is not supported) and clarify that `counter_stop()` stops and resets. With that approach `counter_start()` means just starting from the current state (resuming from pausing or starting from off).

ajf58, 2024-10-09:
Sounds like a good approach to me.

sylvioalves, 2024-10-09:
I see the same issue on ESP32 SoC. `test_valid_function_without_alarm` is currently failing as `counter_start()` is called without any alarm proper configured.

sylvioalves, 2024-10-10:
@ajf58 I submitted that PR that fixes the no_alarm use case. It also fixes an issue with esp32 counter. PTAL.

ajf58, 2024-10-10:
Thanks, @sylvioalves . With your change to the test applied locally (https://github.com/zephyrproject-rtos/zephyr/pull/79631/commits/4527596a23b9ca3271b9f46c07d2aa5ca1e830fd) , this passes with the Raspberry Pi Pico 2 (based on a RP2350A) which shares the same peripheral as the RP2040.

PR #79631 (2024-10-16): tests: counter: add current counter value into no_alarm test
  In the test scenario without alarms, it might
  be the case that counter is already running and
  it will not match the expected tick.
  This adds a tick offset into the expected value
  based on current counter reading.

Could fix #79574 considering the scenario where `counter_start()` do not reset the timer.
closed 2024-10-16 by henrikbrixandersen