zephyrproject-rtos/zephyr #79586: Incorrect 64-bit Counter Behavior on nRF52840DK and nRF52833DK Devices Using counter_get_value_64()

CienetmarkChen, 2024-10-09:
Hi Zephyr Community,

**Describe the bug**
We conducted hardware counter tests on the nRF52840DK[1] and nRF52833DK[2]. During testing, we found that calling [`counter_get_value_64()`](https://docs.zephyrproject.org/apidoc/latest/group__counter__interface.html#ga2dcd0eb10d49c4e5e5024ed75a8e3e86) on these devices returns an incorrect counter value (it appears to remain static).

**To Reproduce**
1. Select an external clock device (e.g., timer, rtc) from the device tree file of the nRF52840DK or nRF52833DK.
2. In the test code, call [`counter_get_value_64()`](https://docs.zephyrproject.org/apidoc/latest/group__counter__interface.html#ga2dcd0eb10d49c4e5e5024ed75a8e3e86) and pass the external clock device as the argument. 
3. Build and flash the application to the board using the following commands:
```bash
west build -p -b nrf52840dk/nrf52840 <path/file>
west flash
```
4.	Observe the output.

**Expected behavior**
The counter value of the external clock device should vary over time.

**Impact**
This issue could affect applications that rely on the 64-bit counter functionality.

**Logs and console output**
The following screenshot shows that the output value of the counter remains constant.
[image]

**Environment:**
OS: Linux
Toolchain: Zephyr SDK 0.16.8
Commit: 318b4957

**Additional context**
[1] https://docs.nordicsemi.com/bundle/ps_nrf52840/page/keyfeatures_html5.html
[2] https://docs.nordicsemi.com/bundle/ps_nrf52833/page/keyfeatures_html5.html

nordicjm, 2024-10-09:
Where did you start the counter?

CienetmarkChen, 2024-10-23:
Hi nordicjm,

Before calling the function [counter_get_value_64](https://docs.zephyrproject.org/apidoc/latest/group__counter__interface.html#ga2dcd0eb10d49c4e5e5024ed75a8e3e86), we ensured to start the counter by calling
[counter_start](https://docs.zephyrproject.org/apidoc/latest/group__counter__interface.html#ga103e0673e31475adcd173601058c72cd) once.

Further testing on the nrf52840dk board showed that calling [counter_get_value_64](https://docs.zephyrproject.org/apidoc/latest/group__counter__interface.html#ga2dcd0eb10d49c4e5e5024ed75a8e3e86) returns `-ENOTSUP`, as defined in [counter.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/counter.h#L428), indicating unsupported functionality.

Here is our experimental code:
`main.c`:
```c
#include <zephyr/kernel.h>
#include <zephyr/sys_clock.h>
#include <zephyr/drivers/counter.h>
#include <stdio.h>

static const struct device *external_sensor_clock = DEVICE_DT_GET(DT_CHOSEN(zephyr_sensor_clock));

int main(void)
{
	uint64_t cycles = 0;
	uint64_t delta_ns = 0;
	int err;
    
    	counter_start(external_sensor_clock);

	while (true) {
		k_sleep(K_MSEC(1000));

		err = sensor_clock_get_cycles(&cycles);
		if (err) {
			printf("Failed to get sensor clock cycles, error: %d\n", err);
		}

		printf("Cycles: %llu\n", cycles);

		delta_ns = sensor_clock_cycles_to_ns(cycles);

		printf("Nanoseconds: %llu\n", delta_ns);
	}

	return 0;
}

int sensor_clock_get_cycles(uint64_t *cycles)
{
	__ASSERT_NO_MSG(counter_is_counting_up(external_sensor_clock));

	int rc;
	rc = counter_get_value_64(external_sensor_clock, cycles);

	return rc;
}

uint64_t sensor_clock_cycles_to_ns(uint64_t cycles)
{
	uint32_t freq = counter_get_frequency(external_sensor_clock);

	if (freq == 0) {
		printf("Sensor clock %s has no fixed frequency\n", external_sensor_clock->name);
		return 0;
	}

	return (cycles * NSEC_PER_SEC) / freq;
}

```

`nrf52840dk_nrf52840.overlay`:
```
/ {
	chosen {
		zephyr,sensor-clock = &timer0;
	};
};

&timer0 {
	status = "okay";
};

&rtc0 {
	status = "okay";
};
```
`prj.conf`:
```
CONFIG_SENSOR=y
CONFIG_COUNTER=y
CONFIG_RTC=y
```

nordicjm, 2024-10-23:
Then the function is not supported, this is not a bug