zephyrproject-rtos/zephyr #79589: No network inface found when enabling ethernet with the "Arduino portenta H7 board" and the "Portenta Hat Carrier" shield

El-Bucheron, 2024-10-09:
**Describe the bug**

Hello,
As mentioned in the title, I am having trouble enabling Ethernet on the [Arduino Portenta H7](https://docs.arduino.cc/hardware/portenta-h7/] when using a [Portenta Hat Carrier](https://docs.arduino.cc/hardware/portenta-hat-carrier/), which includes an Ethernet port. My initial goal was to enable Ethernet, connect the board to my computer, and perform a simple ping test. However, each time I boot the board, the logs display:
`[00:00:00.010,000] <err> net_if: There is no network interface to work with!`
despite my belief that I have configured the Ethernet layer correctly.

I am confident the issue is not related to my Ethernet cable or network, as I have successfully received UDP packets from my computer to the board using an Arduino example code. I also doubt that the problem lies in my code, as it is just a simple k_sleep(K_FOREVER);. Additionally, my prj.conf file is straightforward, with basic settings to enable Ethernet, assign an IP address, and enable logging and the network shell (as detailed in the **To Reproduce** section).

I suspect the issue could be related to the fact that the shield I’m using may not be fully supported by Zephyr. However, after reviewing the schematic of the Portenta Hat Carrier (see the image below), the pin configuration does not appear to have been altered, which leads me to believe the shield should be compatible with the Ethernet implementation.

[image]

Therefore, I believe the problem lies in how the Ethernet configuration is initialized for the Arduino Portenta H7 in the source code—possibly due to incorrect pin configuration or something similar. I would appreciate any assistance or advice. Thank you in advance for your help!

---

**To Reproduce**

Here is the structure of my project "test_ethernet" and the content of the files
```
test_ethernet
├── src
│   ├── main.c
├── app.overlay
├── CMakeLists.txt
└── prj.conf
```
- main.c 
```
#include <zephyr/kernel.h>

int main(void)
{
	/* This sample uses the shell */
	k_sleep(K_FOREVER);

	return 0;
}
```
- app.overlay : empty
- CMakeLists.txt
```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.20.0)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(test_ethernet)

target_sources(app PRIVATE src/main.c)
```
- prf.config
```
# Enabling networking and Ethernet
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_NETWORKING=y
CONFIG_NET_DRIVERS=y
CONFIG_NET_L2_ETHERNET=y

# IPv4 Configuration
CONFIG_NET_IPV4=y
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.56.2"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"

# Logs
CONFIG_NET_LOG=y
CONFIG_LOG=y
CONFIG_NET_STATISTICS=y
CONFIG_PRINTK=y

# Network shell
CONFIG_NET_SHELL=y
CONFIG_SHELL=y
```
---

**Expected behavior**
Something like that (obtained by simulating a qemu board)

```
uart:~$ net iface
Hostname zephyr

Interface eth0 (0x111111) (Ethernet) [1]
======================
.
# Interface Description
.
```

---

**Logs and console output**

Here are the logs when the board boots
```
[00:00:00.010,000] <err> net_if: There is no network interface to work with!
*** Booting Zephyr OS build v3.7.0-2392-g8c32a82e47f5 ***
[00:00:00.010,000] <wrn> net_config: No auto-started network interface - network-bound app initialization skipped.
```

---

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain: Zephyr SDK 0.16.8

nordicjm, 2024-10-09:
@rahul-arasikere can you assist?

El-Bucheron, 2024-10-10:
Hello, @rahul-arasikere, thank you for your response.

Based on your recommendations, I have replaced the [actual mac definition](https://github.com/zephyrproject-rtos/zephyr/blob/66ff30efb4d786ac15a4ef03a81bbe7a3f315690/boards/arduino/portenta_h7/arduino_portenta_h7-common.dtsi#L212C1-L224C1) for the portenta with the one from the board [Nucleo H743ZI](https://github.com/zephyrproject-rtos/zephyr/blob/66ff30efb4d786ac15a4ef03a81bbe7a3f315690/boards/st/nucleo_h743zi/nucleo_h743zi.dts#L196C1-L218C3). Indeed, they use the same microprocessor and the Nucleo H743ZI code includes a MDIO definition and has the "okay" status for both MAC and MDIO. It allowed me to enable the Ethernet interface, as shown in the following log :
```
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131

*** Booting Zephyr OS build v3.7.0-2392-g8c32a82e47f5 ***
[00:00:00.061,000] <inf> net_config: Initializing network
[00:00:00.061,000] <inf> net_config: Waiting interface 1 (0x240015b8) to be up...
[00:00:03.954,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex

[00:00:03.968,000] <inf> net_config: Interface 1 (0x240015b8) coming up
[00:00:03.968,000] <inf> net_config: IPv4 address: 192.168.56.2
``` 
First of all, thank you for that!

However, the Arduino Board still cannot seem to be able to send data through the Ethernet port. When pinging the board from my computer, the board doesn't respond to the ARP request. Similarly, when pinging my computer from my board, no ARP request is sent over the network. Nevertheless, it does seem to be able to receive data, as the board ARP-cache stores the MAC address and the IP of my computer after pinging the board. As explained in the initial commit, I have successfully sent and received UDP packets from the board through a Arduino demo. Thus, I believe this issue is not due to the Ethernet cable, network configuration or firewall. 

Additionally, after modifying of the MAC definition, I noticed the new following CMake Warning related to net drivers when building the project. I am not sure if this is critical for enabling data transmission over Ethernet.
```
CMake Warning at C:/Users/manducher/zephyrproject/zephyr/CMakeLists.txt:952 (message):
  No SOURCES given to Zephyr library: drivers__net
```

Since I do not know what I could next to fix this problem, I am once again asking for help. I apologize if I missed something obvious. I am new to zephyr and embedded systems in general. I can identify the source of the problem but I am not yet skilled enough to fix it myself, especially when it comes to hardware problems. I would greatly appreciate if you could assist me again in solving this new issue.

rahul-arasikere, 2024-10-10:
Some tips since I do not have access to your source code.
I recommend going through the [devicetree troubleshooting guide](https://docs.zephyrproject.org/latest/build/dts/troubleshooting.html#dt-trouble). I do not know is the the two stm32 chips are pin to pin compatible, I usually confirm the pinmuxing against schematics and the stm32cubemx device configurator.
Usually both the nodes are enabled with `status="okay";` in a board overlay file defined for your app.

El-Bucheron, 2024-10-11:
Hello,

Regarding the source code, it is the default and up-to-date version of Zephyr, with the only modification being the MAC definition of the Portenta board, as detailed above. As for the project itself, it is only meant to enable pinging between my board and my computer. The file structure and the content of those files are detailed in my initial message, and simply consists in enabling Ethernet, thenet shell and assigning an IP address.

Following your advice, I will review the troubleshooting guide, and attempt to resolve the issue on my own. If I’m successful, I’ll share the solution and close the issue. Otherwise, I’ll reach out again for further assistance. Thank you once more for your advice, and have a great day.

jukkar, 2024-10-11:
The message
`[00:00:00.010,000] <err> net_if: There is no network interface to work with!`
means there is 0 network interfaces found in the system.
This means that the Ethernet driver was not enabled, or the driver did not create the network interface.

El-Bucheron, 2024-10-11:
Hello @jukkar, thank you for your response. 

With the help of @rahul-arasikere, I was able to modify the configuration file and successfully enable Ethernet, allowing me to perform a simple ping, as shown in the logs below:
```
net ping 192.168.56.100
PING 192.168.56.100
28 bytes from 192.168.56.100 to 192.168.56.2: icmp_seq=1 ttl=128 time=0 ms
28 bytes from 192.168.56.100 to 192.168.56.2: icmp_seq=2 ttl=128 time=0 ms
28 bytes from 192.168.56.100 to 192.168.56.2: icmp_seq=3 ttl=128 time=0 ms
```

I will now close this topic and share my adjustments in case others encounter the same issue. I replaced the existing MAC definition, available [here](https://github.com/zephyrproject-rtos/zephyr/blob/66ff30efb4d786ac15a4ef03a81bbe7a3f315690/boards/arduino/portenta_h7/arduino_portenta_h7-common.dtsi#L212C1-L224C1), with the following configuration:
```
&mac {
	status = "okay";
	pinctrl-0 = < &eth_ref_clk_pa1
		      &eth_crs_dv_pa7
		      &eth_rxd0_pc4
		      &eth_rxd1_pc5
		      &eth_tx_en_pg11
		      &eth_txd1_pg12
		      &eth_txd0_pg13 >;
	pinctrl-names = "default";
};

&mdio {
	status = "okay";
	pinctrl-0 = <&eth_mdio_pa2 &eth_mdc_pc1>;
	pinctrl-names = "default";

	ethernet-phy@0 {
		compatible = "ethernet-phy";
		reg = <0x00>;
		status = "okay";
	};
};
```
This essentially adds an MDIO definition and removes its pin from the MAC definition. However, I am unsure if this configuration would work without the Portenta Hat Carrier, as I have not tested it in that setup.

Thank you again for your help, and have a great day!

closed 2024-10-11 by El-Bucheron