zephyrproject-rtos/zephyr #79594: tests/subsys/debug/coredump* fail to build for mps2/an521/cpu0 error

aescolar, 2024-10-09:
**Describe the bug**
tests/subsys/debug/coredump* fail to build for mps2/an521/cpu0 error

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mps2/an521/cpu0 -GNinja ../tests/subsys/debug/coredump_threads/
3. ninja
4. See error

**Expected behavior**
No build failures

**Impact**

**Logs and console output**
``` 
In file included from /__w/zephyr/zephyr/arch/arm/include/kernel_arch_data.h:29,
                 from /__w/zephyr/zephyr/arch/arm/core/offsets/offsets_aarch32.c:29,
                 from /__w/zephyr/zephyr/arch/arm/core/offsets/offsets.c:9:
/__w/zephyr/zephyr/arch/arm/include/cortex_m/exception.h: In function 'z_arm_set_fault_sp':
/__w/zephyr/zephyr/arch/arm/include/cortex_m/exception.h:261:32: error: 'SCB_CCR_STKALIGN_Msk' undeclared (first use in this function); did you mean 'SCB_CCR_BFHFNMIGN_Msk'?
  261 |         if ((esf->basic.xpsr & SCB_CCR_STKALIGN_Msk) == SCB_CCR_STKALIGN_Msk) {
      |                                ^~~~~~~~~~~~~~~~~~~~
      |                                SCB_CCR_BFHFNMIGN_Msk
/__w/zephyr/zephyr/arch/arm/include/cortex_m/exception.h:261:32: note: each undeclared identifier is reported only once for each function it appears in
``` 
https://github.com/zephyrproject-rtos/zephyr/actions/runs/11250552333/job/31279885750#step:11:1033

**Environment (please complete the following information):**

 - OS: Ubuntu 22.04 CI
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used : main since ec7943bb181753c9b7dc339090a2e7f4b9d7f06e

**Additional context**
Introduced by https://github.com/zephyrproject-rtos/zephyr/pull/73189

hakehuang, 2024-10-09:
nxp  has the same problem

aescolar, 2024-11-12:
Issue reappeared in main tests 
https://github.com/zephyrproject-rtos/zephyr/runs/32781331757
Between others in 
tests/drivers/coredump/coredump_api/debug.coredump.drivers.api (gd32l233r_eval/gd32l233:tests/drivers/coredump/coredump_api/debug.coredump.drivers.api) with error 
```
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DGD32L233 -DIRC32K_VALUE=32000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DTC_RUNID=852eb1bf50d094b07761725068c043b6 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/__w/zephyr/zephyr/kernel/include -I/__w/zephyr/zephyr/arch/arm/include -I/__w/zephyr/zephyr/twister-out/gd32l233r_eval_gd32l233/tests/drivers/coredump/coredump_api/debug.coredump.drivers.api/zephyr/include/generated/zephyr -I/__w/zephyr/zephyr/include -I/__w/zephyr/zephyr/twister-out/gd32l233r_eval_gd32l233/tests/drivers/coredump/coredump_api/debug.coredump.drivers.api/zephyr/include/generated -I/__w/zephyr/zephyr/soc/gd/gd32 -I/__w/zephyr/zephyr/soc/gd/gd32/common/. -I/__w/zephyr/zephyr/soc/gd/gd32/gd32l23x/. -I/__w/zephyr/zephyr/subsys/testsuite/include -I/__w/zephyr/zephyr/subsys/testsuite/coverage -I/__w/zephyr/zephyr/subsys/testsuite/ztest/include -I/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zephyr/zephyr/modules/cmsis/. -I/__w/zephyr/modules/hal/gigadevice/gd32l23x/cmsis/gd/gd32l23x/include -I/__w/zephyr/modules/hal/gigadevice/gd32l23x/standard_peripheral/include -I/__w/zephyr/modules/hal/gigadevice/include -I/__w/zephyr/modules/hal/gigadevice/common_include -isystem /__w/zephyr/zephyr/lib/libc/common/include -fno-strict-aliasing -Werror -Os -imacros /__w/zephyr/zephyr/twister-out/gd32l233r_eval_gd32l233/tests/drivers/coredump/coredump_api/debug.coredump.drivers.api/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m23 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /__w/zephyr/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/zephyr/zephyr/tests/drivers/coredump/coredump_api=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -fno-lto -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -c /__w/zephyr/zephyr/arch/arm/core/offsets/offsets.c
In file included from /__w/zephyr/zephyr/arch/arm/include/kernel_arch_data.h:29,
                 from /__w/zephyr/zephyr/arch/arm/core/offsets/offsets_aarch32.c:29,
                 from /__w/zephyr/zephyr/arch/arm/core/offsets/offsets.c:9:
/__w/zephyr/zephyr/arch/arm/include/cortex_m/exception.h: In function 'z_arm_set_fault_sp':
/__w/zephyr/zephyr/arch/arm/include/cortex_m/exception.h:262:32: error: 'SCB_CCR_STKALIGN_Msk' undeclared (first use in this function); did you mean 'SCB_CCR_BFHFNMIGN_Msk'?
  262 |         if ((esf->basic.xpsr & SCB_CCR_STKALIGN_Msk) == SCB_CCR_STKALIGN_Msk) {
      |                                ^~~~~~~~~~~~~~~~~~~~
      |                                SCB_CCR_BFHFNMIGN_Msk
/__w/zephyr/zephyr/arch/arm/include/cortex_m/exception.h:262:32: note: each undeclared identifier is reported only once for each function it appears in
```

aescolar, 2024-11-12:
@ithinuel @mrkhldn ^^

ithinuel, 2024-11-13:
Ah, right, this flag (STKALIGN) is absent for both base and main line but the code currently only excluded the snippet if it is building for v8m-main.

PR #79599 (2024-10-09): Revert "coredump: ARM: Ensure sp in dump is set as gdb expects"
This reverts commit ec7943bb181753c9b7dc339090a2e7f4b9d7f06e.
This commit introduced a regression.
Let's revert it so we do not block development in main.
For more information see:
https://github.com/zephyrproject-rtos/zephyr/issues/79594

Fixes #79594
closed 2024-11-14 by mmahadevan108