zephyrproject-rtos/zephyr #79607: mcuboot is failing with stm32l486zg, when ram is set to 128k

tt0776, 2024-10-09:
mcuboot is failing with stm32l486zg, when ram is set to 128k

The failing is specific to when including the SRAM 32kb with hardware parity check in dts/arm/st/l4/stm32l486Xg.dtsi
[reg = <0x20000000 DT_SIZE_K(128)>;](https://github.com/zephyrproject-rtos/zephyr/blob/1726443d9d2584bc6adb7fd08d63a5fc645ce09a/dts/arm/st/l4/stm32l486Xg.dtsi#L12)

mcuboot works correct, when excluding the the 32kb and using the one defined in dts/arm/st/l4/stm32l476Xg.dtsi
[reg = <0x20000000 DT_SIZE_K(96)>;](https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/st/l4/stm32l476Xg.dtsi#L11)

FRASTM, 2024-10-09:
On the stm32l47x/l48x devices, the SRAM2 is located at 0x1000 0000
There is no continous mapping with SRAM1 (0x2000 0000)   
Could this explain the pb you see
That mapping differs from stm32l49x/L4ax

tt0776, 2024-10-10:
Sure @FRASTM that sound likely the reason, haven't look in to this details.
Just start of the datasheet tells me that the 32kb was not the same as the 96kb.
When reducing the memory to the 96kb, everything works in mcuboot.
I think when the stm32l476Xg.dtsi was added, there was just looked at the 128kb and changed to that amount.

From the ST datasheet DS10199 Rev 8 19/258
```
STM32L486xx devices feature 128 Kbyte of embedded SRAM, split into two blocks:
• 96 Kbyte mapped at address 0x2000 0000 (SRAM1)
• 32 Kbyte located at address 0x1000 0000 with hardware parity check (SRAM2).
This block is accessed through the ICode/DCode buses for maximum performance.
These 32 Kbyte SRAM can also be retained in Standby mode.
The SRAM2 can be write-protected with 1 Kbyte granularity.
The memory can be accessed in read/write at CPU clock speed with 0 wait states.
```

I am not sure that the feature has a driver in zephyr that match use of this SRAM2 range correctly?

tt0776, 2024-10-10:
@FRASTM thanks for you help with support for microchip requires_ulbpr 
The driver works with the microchip flash and ST driver, I have added some warning from the actual compile, to the closed merged   ticket.
[drivers: flash: qspi stm32 driver supporting requires_ulbpr ](https://github.com/zephyrproject-rtos/zephyr/pull/74534#issuecomment-2388530893)
Not sure how to address this forward and getting feedback on possible fix to warnings.
Do I open a new ticket or keep the comment on the closed ticket ?

erwango, 2024-10-10:
> Do I open a new ticket or keep the comment on the closed ticket ?

Yes please open a new ticket

tt0776, 2024-10-14:
@erwango it look right this can be a fix, I will make an attempt to create a new PR, when done testing with a overlay.
This this the page to check for doing a PR or is there another quick guide?
[Contribution Guidelines](https://docs.zephyrproject.org/latest/contribute/guidelines.html)

tt0776, 2024-10-14:
@erwango I tested with a overlay and the mcuboot don't fail after download of new firmware any longer.

The only thing is that the program mapping don't report more or less ram depending on "sram1" is included?

"sram1" It is pressent in ... /zephyr/include/generated/zephyr/devicetree_generated.h when having it in the board overlay
When removing it from overlay it is not to be found again in devicetree_generated.h. 
It may or may not actually be usable, is there a way of testing it out with a sample on my board?

Example after a compile of modified mcuboot sample
With patch `sram1: memory@10000000`
```
[12/12] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      105860 B       818 KB     12.64%
             RAM:       29008 B        96 KB     29.51%
        IDT_LIST:          0 GB        32 KB      0.00%
```

Without patch `sram1`
```
[209/209] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      105860 B       818 KB     12.64%
             RAM:       29008 B        96 KB     29.51%
        IDT_LIST:          0 GB        32 KB      0.00%
```

erwango, 2024-11-06:
I'll open a PR to fix.

closed 2024-11-22 by kartben