zephyrproject-rtos/zephyr #79623: ESP32 SOCs can't be fully debugged using openocd

Szewcson, 2024-10-09:
**Describe the bug**
When building appimage for esp32 soc's it is lacking partition table by default. Openocd using partition table to determine where it should store software breakpoints. When there is no partition table only 2 hardware breakpoints are available. There is an option to pass appimage_offset to openocd as an workaround but default built is not using an offset as I understand.  There is also an option to pass binary partition table to a runner, but I can't find info how to generate one for zephyr. In that case I think partition table should be build and included by default, or docs should be updated to allow inexperienced users debugging without to much dinging in expert level esp-idf and zephyr documentation.

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Debian Bookworm
 - Zephyr SDK 0.16.8, openocd-esp32 (Open On-Chip Debugger v0.12.0-esp32-20240821-85-gad4f1253 (2024-10-08-14:11) - built from master branch)
 - Zephyr 3.7.0

Szewcson, 2024-10-12:
I tried to set appimage_offset in openocd to values that are printed during boot bout with no luck.
`ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x2c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40800000,len:0x83dc
load:0x408083f0,len:0x16b4
SHA-256 comparison failed:
Calculated: f3a2dcd05a5e41fe8422a54a06e1f61c91cbb52c04b8a90b1ccec59b0f50d780
Expected: 0000000030650000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40801782
I (46) boot: ESP Simple boot
I (46) boot: compile time Oct 12 2024 15:28:08
I (46) spi_flash: detected chip: gd
I (47) spi_flash: flash io: dio
W (48) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (60) boot: chip revision: v0.0
I (63) boot.esp32c6: SPI Speed      : 80MHz
I (66) boot.esp32c6: SPI Mode       : DIO
I (70) boot.esp32c6: SPI Flash Size : 8MB
I (74) boot: Enabling RNG early entropy source...
I (79) boot: DRAM: lma 0x00000020 vma 0x40800000 len 0x83dc   (33756)
I (85) boot: DRAM: lma 0x00008404 vma 0x408083f0 len 0x16b4   (5812)
I (91) boot: padd: lma 0x00009ac8 vma 0x00000000 len 0x6530   (25904)
I (97) boot: IMAP: lma 0x00010000 vma 0x42010000 len 0xa74    (2676)
I (103) boot: padd: lma 0x00010a7c vma 0x00000000 len 0xf57c   (62844)
I (109) boot: IMAP: lma 0x00020000 vma 0x42000000 len 0x3aac   (15020)
I (115) boot: Image with 6 segments
I (119) boot: DROM segment: paddr=00010000h, vaddr=42010000h, size=00A80h (  2688) map
I (126) boot: IROM segment: paddr=00020000h, vaddr=42000000h, size=03AACh ( 15020) map

I (169) heap: ESP runtime heap init at 0x4080c350 size 448 kB.

*** Booting Zephyr OS build v3.7.0-4288-g4793e00ae803 ***
You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:
Start typing characters to see them echoed back`

both 0x00020000 and 0x00010000 produced the same warnings.

and, yes `W (48) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.` - but it is some issue with esp32c6 and esp-idf and without zephyr I have that one  but no warnings about partition table. I also tried changing  - --esp-app-address in runners.yaml but also with no luck.

closed 2024-12-30 by github-actions[bot]

EricNRS, 2024-12-30:
Getting software breakpoints working would help immensely for debugging efforts, so would be good to find out if there is an appropriate work-around.  I currently add inline assembly breakpoints as a work-around, but this requires recompiling the code.

@erhankur - you are the expert in this area, any ideas on how to get software breakpoints working even if it is a manual work-around?

erhankur, 2024-12-31:
@EricNRS please try this branch

https://github.com/erhankur/openocd-esp32/tree/add_flash_map_manually

When you start the OpenOCD , you will need to set DROM, IROM mapping manually,

For example; 
```
[esp32] [INF] Application start=40082b34h
[esp32] [INF] DRAM segment: paddr=00024898h, vaddr=3ffb0000h, size=0053ch (  1340) load
[esp32] [INF] IRAM segment: paddr=00020040h, vaddr=40080000h, size=04858h ( 18520) load
I (125) boot: DROM segment: paddr=00030000h, vaddr=3f400000h, size=01030h (  4144) map
I (125) boot: IROM segment: paddr=00040000h, vaddr=400d0000h, size=03238h ( 12856) map
```

```
openocd  -c 'set ESP_RTOS Zephyr; set ESP_ONLYCPU 1' -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "esp flash_map 0x30000 0x3f400000 0x1030 0x40000 0x400d0000 0x3238 0 0 0x800000"
```

`esp flash_map` command usage:
```
<drom_paddr><drom_vaddr><drom_size><irom_paddr><irom_vaddr><irom_size><flash_paddr><flash_vaddr><flash_size>
```
`flash_paddr` and `flash_vaddr` will be zero. `flash_size` will be the total size of flash.

If you can't build, you can get the artifcats from https://github.com/erhankur/openocd-esp32/actions/runs/12558340623

Szewcson, 2025-01-06:
@erhankur how can I use esp flash_map command in vscode with cortex-debug plugin?

EricNRS, 2025-01-06:
> how can I use esp flash_map command in vscode with cortex-debug plugin?
That will not work.  Cortex-debug is for ARM Cortex processors and ESP32 chips are either Xtensa or RISC-V.

Szewcson, 2025-01-06:
So how can I integrate it in vscode?

sylvioalves, 2025-01-06:
@Szewcson, would you mind checking this repo?
https://github.com/LucasTambor/EspZephyrWS

Szewcson, 2025-01-06:
> @Szewcson I don't have VSCode setup for now but you should be able to set it after specifing the config file. -c command order matters and must be the same as above

Ok, I was able to hack it. I added those commands in zephyr/boards/espressif/esb32c6_devkitc/support/openocd.cfg and use that file from my launch.json cortex-debug setup. Also commented out openOCDPreConfigLaunchCommands section since it should do the same but I can't get right order by using it. THX. @sylvioalves  I will take a look at it.

erhankur, 2025-01-21:
@Szewcson @EricNRS Currently, in the master branch, we have support for both simple boot and MCU boot mappings. You no longer need to specify the entire mapping manually. Setting the `appimage_offset` is mandatory, and it will be enough.

EricNRS, 2025-01-21:
> [@Szewcson](https://github.com/Szewcson) [@EricNRS](https://github.com/EricNRS) Currently, in the master branch, we have support for both simple boot and MCU boot mappings. You no longer need to specify the entire mapping manually. Setting the `appimage_offset` is mandatory, and it will be enough.

Awesome, thank you Erhan!  I'll do a new build of the Espressif OpenOCD and give it a try.

I'll give Lucas's .svd files a try too with Cortex Debug . . . quite funny to have a .svd file for a non-ARM part.