zephyrproject-rtos/zephyr #79627: tests: kernel: thread_apis: test fails on qemu_cortex_r5 because assumed invalid address is valid

ibirnbaum, 2024-10-09:
With MPU and userspace enabled on qemu_cortex_r5, the test case test_thread_name_user_get_set erroneously fails constantly, as the function k_thread_name_set is called with a fixed str pointer of 0xFFFFFFF0, assumed to be inaccessible memory. On qemu_cortex_r5, this address is valid (top end of the ZynqMP's peripherals register space).

Changing the address to an actually inaccessible address will make the test case pass.

ibirnbaum, 2024-10-09:
Would it be possible to have some sort of override parameter for tests where invalid memory is accessed deliberately, or would that be exaggerated in case that only this one test case is affected by the described effect?

peter-mitsis, 2025-01-09:
@ibirnbaum - I have an idea how to resolve this. Essentially, it boils down to creating a Kconfig option for the test that specifies the invalid address to use. For platforms where that default value does not work, we create a conf overlay file that specifies a value that would work for it. What would be a good value for your case on the qemu_cortex_r5?

ibirnbaum, 2025-01-13:
@peter-mitsis Sounds good! In the 32-bit address space, the Ultrascale has two reserved, currently unused address ranges:

- 0xF0000000 - 0xF7FFFFFF
- 0xF9200000 - 0xFCFFFFFF

Those should be a pretty safe bet for an invalid memory access, as the MPU setup will never cover those areas.

PR #83932 (2025-01-16): tests: thread_apis: Allow configurable unmapped address
Although many platforms may have an unmapped memory address in common, this address can not be expected to be the same for all platforms. To that end, the thread_apis test has been updated to use a configurable value for the unmapped memory address in _test_thread_name_user_get_set_.

Fixes #79627
commit 1e87400b (2025-01-16): tests: thread_apis: Add custom unmapped address

Adds a custom unmapped address for qemu_cortex_r5.

Fixes #79627
closed 2025-01-16 by kartben