zephyrproject-rtos/zephyr #79654: Warning when using flash_stm32_qspi with CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32 and requires-ulbpr

tt0776, 2024-10-10:
Feedback and question on warning

What is not correct configured or how to possible fix in zephyr/drivers/flash/flash_stm32_qspi.c?
Driver looks to work correct with sample and able to read and write.

I have checked out main and tested when microchip patch was added(#74534) and used it to flash for microchip sst26vf032b with sample "samples/drivers/spi_flash"
For easier reproduction warning, I have attached overlay for "stm32l476g_disco" to be placed in sample "boards"
 (Need rename to .conf and .overlay to use) 
[stm32l476g_disco_conf.txt](https://github.com/user-attachments/files/17239807/stm32l476g_disco_conf.txt)
[stm32l476g_disco_overlay.txt](https://github.com/user-attachments/files/17239808/stm32l476g_disco_overlay.txt)

Specific enable and using.
`CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32=y`
`requires-ulbpr;`

Example on location of config[flash_stm32_qspi.c "CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32" and "requires-ulbpr "](https://github.com/zephyrproject-rtos/zephyr/blob/1726443d9d2584bc6adb7fd08d63a5fc645ce09a/drivers/flash/flash_stm32_qspi.c#L190)

I get warning on this and not sure on how to fix it or it will be a problem.
`Warning (spi_bus_bridge): /soc/quadspi@a0001000: node name for SPI buses`
`Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'`

Example on "my_board" as derived from nucleo_l476rg

### relevant devicetree part for [sst26vf032b](https://ww1.microchip.com/downloads/en/devicedoc/20005218e.pdf)
```
sst26vf032b: qspi-nor-flash@90000000 {
		// https://docs.zephyrproject.org/latest/build/dts/api/bindings/flash_controller/st,stm32-qspi-nor.html
		compatible = "st,stm32-qspi-nor";
		reg = <0x90000000 DT_SIZE_M(32)>; // 4 MBytes
		qspi-max-frequency = <DT_FREQ_M(80)>;
		status = "okay"; // Is only to be enable when actually using external flash.
		//NOT IN USE// reset-gpios = <&gpiod 3 GPIO_ACTIVE_LOW>;
        //NOT IN USE// reset-gpios-duration = <1>; // The duration (in ms) for the flash memory reset pulse
		spi-bus-width = <4>;
		requires-ulbpr;		
		page-size = <4096>; // Force “Erase 4 Kbytes” (0x20) to insure using smalles possible page size possible to support
		writeoc = "PP_1_1_4";

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "nor";
				reg = <0x00000000 DT_SIZE_M(4)>;
			};
		};
	};
};
```

### Full printout with some path substitutes 

```
COMMAND: west build -p always /home/ ... /samples/drivers/spi_flash -d /home/ ... /build -b my_board@3.0.1  -- -DBOARD_ROOT=/home/ ... /pjt
-- west build: making build dir /home/ ... /build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ ... /samples/drivers/spi_flash
-- CMake version: 3.30.2
-- Found Python3: /home/ ... /zephyrproject/.venv/bin/python3 (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/ ... /.cache/zephyr
-- Zephyr version: 3.7.99 (/home/ ... /zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: my_board, Revision: 3.0.1, qualifiers: stm32l476xx
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/ ... /zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/ ... /zephyr-sdk-0.16.8)
-- Found Dtc: /home/ ... /zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")

-- Including generated dts.cmake file: / ... /zephyr/dts.cmake
CMake Warning at /home/ ... /zephyrproject/zephyr/cmake/modules/dts.cmake:405 (message):
  dtc raised one or more warnings:

  /home/ ... / ... /zephyr/zephyr.dts:446.29-485.5:
  Warning (spi_bus_bridge): /soc/quadspi@a0001000: node name for SPI buses
  should be 'spi'

  <stdout>: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'

Call Stack (most recent call first):
  /home/ ... /zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /home/ ... /zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/ ... /zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

...

'/home/ ... / ... /zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/ ... /zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ ... /zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Setting build type to 'MinSizeRel' as none was specified.
-- Using ccache: /usr/bin/ccache
-- Configuring done (7.8s)
-- Generating done (0.3s)
-- Build files have been written to: /home/ ... /build
-- west build: building application
[1/165] Preparing syscall dependency handling

[3/165] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (/home/ ... /zephyrproject/zephyr), build: v3.7.0-3720-g9c0f92db475f
[165/165] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       42444 B         1 MB      4.05%
             RAM:        5888 B        96 KB      5.99%
        IDT_LIST:          0 GB        32 KB      0.00%

```

FRASTM, 2024-10-14:
Not sure the reason for that warning, but adding pre_dt_board.cmake to the avoid that DTS message is going to remove the pb

```
# SPI is implemented via octospi so node name isn't spi@...
list(APPEND EXTRA_DTC_FLAGS "-Wno-spi_bus_bridge")
```

tt0776, 2024-10-14:
@FRASTM thanks it looks to work for my board

FRASTM, 2024-10-16:
> @FRASTM thanks it looks to work for my board

Should also with the new version of the related PR

tt0776, 2025-01-13:
@mathieuchopstm  is there anything I need to do to get this merged in 3.7 LTS ?

I still needs to cherry pick this 3 commits for the newer LTS 3.7.1 

`git cherry-pick -n 72370b23ce643f02bc57c5d005f9d0e324f92e9b 76740ae1d12e8a436c33e8f9b92d9ec0e5f40263 ff34d575bc154f79271ecfb79c0362ef9e9cd73f`

---------------------------------------------------------------------
dts: bindings: flash_controller stm32 qspi has requires-ulbpr property

Add the <requires-ulbpr> property from the "jedec,spi-nor-common.yaml"
to the existing st,stm32-qspi-nor.yaml. So that external quad-NOR with
unlock the Global Block Protection (BPR) (opcode 0x98) is accepted.

SHA: 72370b23ce643f02bc57c5d005f9d0e324f92e9b
 
---------------------------------------------------------------------
drivers: flash: qspi stm32 driver supporting requires_ulbpr

Add the support of the requires_ulbpr property when a Microchip
quad-spi flash is mounted.
Set the CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32=y flag to access
the command

SHA: 76740ae1d12e8a436c33e8f9b92d9ec0e5f40263

---------------------------------------------------------------------
drivers: flash: stm32 qspi flash driver with unprotect command

Add the write_unprotect command to the stm32 qspi flash driver
to un protect flash before any write operation to the external quad-NOR

SHA: ff34d575bc154f79271ecfb79c0362ef9e9cd73f

tt0776, 2025-01-13:
@mathieuchopstm as a note to this 3 commits, it looks like the main branch got them reversed.

The cherry-pick worked Friday jan. 10 and here Monday jan 13, it is not possible to apply anylonger?

I needed to patch it then in this way:

```
diff --git a/drivers/flash/Kconfig.stm32 b/drivers/flash/Kconfig.stm32
index 9fa3e8e555..893e54f10f 100644
--- a/drivers/flash/Kconfig.stm32
+++ b/drivers/flash/Kconfig.stm32
@@ -83,4 +83,12 @@ config FLASH_STM32_BLOCK_REGISTERS
 	  registers improves system security, because flash content (or
 	  protection settings) can't be changed even when exploit was found.
 
+config USE_MICROCHIP_QSPI_FLASH_WITH_STM32
+	bool "Include patch for Microchip qspi flash when running with stm32"
+	depends on DT_HAS_ST_STM32_QSPI_NOR_ENABLED
+	help
+	  Set to use Microchip qspi flash memories which supports
+	  the Global Block Protection Unlock instruction (ULBPR - 98H),
+	  and write with SPI_NOR_CMD_PP_1_1_4 on 4 lines
+
 endif # SOC_FLASH_STM32
diff --git a/drivers/flash/flash_stm32_qspi.c b/drivers/flash/flash_stm32_qspi.c
index 0ff33baf55..a66b49e88b 100644
--- a/drivers/flash/flash_stm32_qspi.c
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -184,9 +184,16 @@ static inline int qspi_prepare_quad_program(const struct device *dev,
 			dev_data->qspi_write_cmd == SPI_NOR_CMD_PP_1_4_4);
 
 	cmd->Instruction = dev_data->qspi_write_cmd;
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+	/* Microchip qspi-NOR flash, does not follow the standard rules */
+	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4) {
+		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
+	}
+#else
 	cmd->AddressMode = ((cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
 				? QSPI_ADDRESS_1_LINE
 				: QSPI_ADDRESS_4_LINES);
+#endif /* CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32 */
 	cmd->DataMode = QSPI_DATA_4_LINES;
 	cmd->DummyCycles = 0;
 
@@ -340,6 +347,27 @@ static int qspi_read_jedec_id(const struct device *dev, uint8_t *id)
 }
 #endif /* CONFIG_FLASH_JESD216_API */
 
+static int qspi_write_unprotect(const struct device *dev)
+{
+	int ret = 0;
+	QSPI_CommandTypeDef cmd_unprotect = {
+			.Instruction = SPI_NOR_CMD_ULBPR,
+			.InstructionMode = QSPI_INSTRUCTION_1_LINE,
+	};
+
+	if (IS_ENABLED(DT_INST_PROP(0, requires_ulbpr))) {
+		ret = qspi_send_cmd(dev, &cmd_write_en);
+
+		if (ret != 0) {
+			return ret;
+		}
+
+		ret = qspi_send_cmd(dev, &cmd_unprotect);
+	}
+
+	return ret;
+}
+
 /*
  * Read Serial Flash Discovery Parameter
  */
@@ -1483,6 +1511,13 @@ static int flash_stm32_qspi_init(const struct device *dev)
 	}
 #endif /* CONFIG_FLASH_PAGE_LAYOUT */
 
+	ret = qspi_write_unprotect(dev);
+	if (ret != 0) {
+		LOG_ERR("write unprotect failed: %d", ret);
+		return -ENODEV;
+	}
+	LOG_DBG("Write Un-protected");
+
 #ifdef CONFIG_STM32_MEMMAP
 #if DT_PROP(DT_NODELABEL(quadspi), dual_flash) && defined(QUADSPI_CR_DFM)
 	/*
diff --git a/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml b/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
index 3f990f4a12..8c23e02c07 100644
--- a/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
+++ b/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
@@ -61,3 +61,13 @@ properties:
       supporting 1-4-4 mode also would support fast page programming.
 
       If absent, then 1-4-4 program page is used in quad mode.
+
+  requires-ulbpr:
+    type: boolean
+    description: |
+      Indicates the device requires the ULBPR (0x98) command.
+
+      Some flash chips such as the Microchip SST26VF series have a block
+      protection register that initializes to write-protected.  Use this
+      property to indicate that the BPR must be unlocked before write
+      operations can proceed.

```

mathieuchopstm, 2025-01-13:
Hi @tt0776,
I merely re-opened the issue because #79797 which fixes the issue you reported (`Warning (spi_bus_bridge): /soc/quadspi@a0001000: node name for SPI buses`) has not been merged yet.

---

As for your request, per https://docs.zephyrproject.org/latest/project/release_process.html#long-term-support-and-maintenance:
```
changes from main branch to an LTS branch will be limited to fixes
that apply to both branches and for existing features only
```
The commits you are referring to appear to be introducing new features; as such, I do not think they would be applicable for a backport to Zephyr 3.7 LTS.

closed 2025-01-17 by kartben