zephyrproject-rtos/zephyr #79660: stm32 rcc reserved AHB3 bit 8 is being set at startup.

GeorgeCGV, 2024-10-10:
**Describe the bug**
A reserved bit at position `8` is being set to `1` in `static int stm32h7_flash_init(const struct device *dev)` in `RCC_AHB3ENR` register:

**RCC_AHB3ENR**
[image]

The clock and bit to set come from `zephyr/dts/arm/st/h7/stm32h7.dtsi`:
```c
		flash: flash-controller@52002000 {
			compatible = "st,stm32-flash-controller", "st,stm32h7-flash-controller";
			reg = <0x52002000 0x400>;
			interrupts = <4 0>;
			clocks = <&rcc STM32_CLOCK_BUS_AHB3 0x00000100>;

			#address-cells = <1>;
			#size-cells = <1>;
		};
```

@erwango maybe the intention was to set `FLASHLPEN` in `RCC_AHB3LPENR`? The bit position matches and register contains `AHB3` in the name.

**RCC_AHB3LPENR**
[image]

**To Reproduce**
1. Build anything that has `st,stm32h7-flash-controller` driver and observe `RCC_AHB3ENR` state.

**Expected behavior**
The reserved bit shall not be set.

**Environment (please complete the following information):**

 - 66ff30efb4d786ac15a4ef03a81bbe7a3f315690

FRASTM, 2024-10-28:
The Bit is actually the 8th of the RCC_AHB3ENR and some stm32h7 devices have not this bit: only stm32H7 dual core devices 
have. Earlier we took the option to set the bit for _all_ stm32h7 else we should have to identify dual core devices inside the stm32h7. We considered at that time that writing the bit 8 of the RCC_AHB3ENR did not cause any wrong behavior when the bit was not defined.

FRASTM, 2024-10-28:
We can now define the clock property only for stm32h7_dual core mcus and   have this clock_on in the stm32h7_flash_init only for the "st,stm32h7-flash-controller" with thatclock property

closed 2024-11-04 by mmahadevan108