zephyrproject-rtos/zephyr #79664: ZBus linking problem with C++ on latest main branch

OgnjenX, 2024-10-10:
I have this issue caused by C++ mangling (I believe):

`ld.bfd.exe: app/libapp.a(TemperatureEvent.cpp.obj):(._zbus_channel_observation.static.temperature_chan00_+0x4): undefined reference to `computing_th_temp_lis'`

Here is the code:

```
ZBUS_CHAN_DEFINE(temperature_chan,
                 TemperatureEvent,
                 nullptr,
                 nullptr,
                 ZBUS_OBSERVERS(computing_th_temp_lis),
                 ZBUS_MSG_INIT()
);
```

and listender is defined in another cpp file:

`ZBUS_LISTENER_DEFINE(computing_th_temp_lis, on_zbus_temperature_callback);`

It works on 3.7 branch.

rodrigopex, 2024-10-11:
@OgnjenX, is that working on Zephyr v3.7.0 and not in the main?

OgnjenX, 2024-10-11:
Yes.

vChavezB, 2024-10-14:
Found the commit that originates the issue

https://github.com/zephyrproject-rtos/zephyr/commit/1ed77cb50ce33cce9c87f88c1374932bf6be85fb

vChavezB, 2024-10-14:
PR that fixes the issue

https://github.com/zephyrproject-rtos/zephyr/pull/79793

closed 2024-10-16 by henrikbrixandersen