zephyrproject-rtos/zephyr #79671: Wrong channel index in connectionless AoA

CaveCanem1240, 2024-10-10:
**Describe the bug**

I am working on direction_finding_connectionless_rx/tx samples using Nordic nrf52833 and nrf5340. I noticed that with `PER_ADV_EVENT_CTE_COUNT` set to `5` on the tx side, CTE reports from the rx side are like 
```
PER_ADV_SYNC[0]: [DEVICE]: EF:BB:1A:FA:87:19 (random) synced, Interval 0x03c0 (1200 ms), PHY LE 2M
success. Periodic sync established.
Enable receiving of CTE...
success. CTE receive enabled.
Scan disable...Success.
Waiting for periodic sync lost...
CTE[0]: samples count 45, channel ID 24, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -690
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -760
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -760
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -760
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC not OK, CTE Info OK, RSSI -760
CTE[0]: samples count 45, channel ID 36, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -410
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -380
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -380
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -380
CTE[0]: samples count 45, channel ID 0, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -380
PER_ADV_SYNC[0]: [DEVICE]: EF:BB:1A:FA:87:19 (random) sync terminated
Periodic sync lost.
```
They are basically a random selected channel `N` followed by four `0` for each CTE train. As far as I know, channel number within a single train should be the same in Zephyr. Please correct me if not.

I also tested it with a ST processor as the TX (not running on Zephyr), with RX remained unchanged, and `PER_ADV_EVENT_CTE_COUNT` was set to `4`. As you can see from the log message below, channel index were 4 different numbers for the ST part (ST has a different channel selecting strategy). That means the `direction_finding_connectionless_rx` sample was working as expected and it was `direction_finding_connectionless_tx` that might have some issue.
```
Waiting for periodic sync lost...
CTE[0]: samples count 45, channel ID 31, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -480
CTE[0]: samples count 45, channel ID 7, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -470
CTE[0]: samples count 45, channel ID 32, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -480
CTE[0]: samples count 45, channel ID 18, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -480
PER_ADV_SYNC[0]: [DEVICE]: A0:AC:7E:00:00:00 (public) sync terminated
```

**To Reproduce**

Zephyr version: 3.7.99
SDK: zephyr-sdk-0.16.8
TX board: nrf52833 devkit
RX board: nrf5340 devkit

For TX side: build `direction_finding_connectionless_tx` and flash
```
west build -p always -b nrf52833dk/nrf52833 samples/bluetooth/direction_finding_connectionless_tx -DEXTRA_CONF_FILE=overlay-aoa.conf
west flash
```
For RX side: 
1. add channel ID print to `cte_recv_cb` in `direction_finding_connectionless_rx`
```
static void cte_recv_cb(struct bt_le_per_adv_sync *sync,
			struct bt_df_per_adv_sync_iq_samples_report const *report)
{
	printk("CTE[%u]: samples count %d, channel ID %u, cte type %s, slot durations: %u [us], "
	       "packet status %s, RSSI %i\n",
	       bt_le_per_adv_sync_get_index(sync), report->sample_count, report->chan_idx,
	       cte_type2str(report->cte_type), report->slot_durations,
	       packet_status2str(report->packet_status), report->rssi);
}
```
2. build and flash

**Impact**

AoA calculation uses channel index for wavelength calculation. There would be bias if incorrect.

**Additional context**

The bug was first found on nRF Connect SDK, then I switched to Zephyr Project and was able to reproduce the problem. There are some details I posted in Nordic Forum https://devzone.nordicsemi.com/f/nordic-q-a/114492/direction-finding-chan_idx-of-cte-sent-in-single-periodic-advertising-train, which I didn't get an answer from Nordic but could be some useful FYI.

CaveCanem1240, 2024-10-10:
Looks like there is a 3-years-old `FIXME` here [https://github.com/zephyrproject-rtos/zephyr/pull/38989/commits/28cadd13356e2228ce8f61e1078dd501584d26dc](https://github.com/zephyrproject-rtos/zephyr/pull/38989/commits/28cadd13356e2228ce8f61e1078dd501584d26dc)

alwa-nordic, 2024-10-11:
@cvinayak Is this a Controller bug?

cvinayak, 2024-10-14:
> Looks like there is a 3-years-old `FIXME` here [28cadd1](https://github.com/zephyrproject-rtos/zephyr/commit/28cadd13356e2228ce8f61e1078dd501584d26dc)

Thanks for digging out, seems like we overlooked here.

CaveCanem1240, 2024-10-14:
Thanks, I appreciate you putting this task onto the schedule!

cvinayak, 2024-10-15:
> Thanks, I appreciate you putting this task onto the schedule!

Please use this fix PR branch and perform your tests, your new results with the plot as in devzone will be valuable validation of the fix. Thank you in advance.

Fix: https://github.com/zephyrproject-rtos/zephyr/pull/79818

CaveCanem1240, 2024-10-15:
> Please use this fix PR branch and perform your tests, your new results with the plot as in devzone will be valuable validation of the fix.

@cvinayak Thanks for the quick fix! Channel indexes within a periodic train are different numbers now. And I test similar physical setup as in devzone, here are some plots after applying the commit:
[image]
The separation issue for big channel index numbers I mentioned in post has gone, which is good, but still larger offset can be observed when channel idx goes up.

cvinayak, 2024-10-15:
Thank you for testing and confirming that now periodic train have different channel indices in the reports. I am not familar with your plots as to the X and Y axis. I will need to let folks at the devzone provide their comments.

PR #79818 (2024-10-18): Bluetooth: Controller: Fix missing chain PDU channel indices
Fix missing population of channel indices for Extended Advertising Chain PDUs.

Fixes #79671.
closed 2024-10-18 by dleach02