zephyrproject-rtos/zephyr #79702: samples: subsys: usb: mass/sample.usb_device_next.mass_ram_none and  cdc_acm/sample.usb_device_next.cdc-acm fails on nrf54h20

nordic-piks, 2024-10-11:
**Describe the bug**
The tests:
- samples/subsys/usb/cdc_acm/sample.usb_device_next.cdc-acm
- samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none

Observed for
- nrf54h20dk/nrf54h20/cpuap

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk/nrf54h20/cpuap connected
2. go to your zephyr dir
3. call ./scripts/twister -T  samples/subsys/usb/ -p nrf54h20dk/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4183-g66ff30efb4d7 ***
[00:00:01.181,586] <err> udc_dwc2: Quirk init failed -11
[00:00:01.181,594] <err> usbd_core: Failed to init device driver
[00:00:01.181,598] <err> usbd_sample_config: Failed to initialize device support
[00:00:01.181,601] <err> cdc_acm_echo: Failed to initialize USB device
[00:00:01.181,604] <err> cdc_acm_echo: Failed to enable USB
```
There are expected ping/pong messages.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.8
 - Commit SHA or Version used: v3.7.0-4183-g66ff30efb4d7

tmon-nordic, 2024-10-11:
@lstnl @Rafal-Nordic This is nrfs issue. Reverting 022122d659a7c10180b8a91c6a0b49874f2850fa is a possible solution. I let you fix it.

nordic-piks, 2024-11-07:
Triaged internally

nordic-piks, 2024-11-26:
Cannot reproduce.

closed 2024-11-26 by nordic-piks