zephyrproject-rtos/zephyr #79709: cmake: string function usage error

hakehuang, 2024-10-11:
- **Description:**

in below file cmake\modules\yaml.cmake
https://github.com/zephyrproject-rtos/zephyr/blob/4793e00ae803ae6b7948e4c110f7ceef11821ef5/cmake/modules/yaml.cmake#L303
if we set the ZEPHYR_SDK_INSTALL_DIR as "gnuarmemb", then we will get below
'''
{
  "cmake": {
    "toolchain": {
      "name": "gnuarmemb"
    }
  }
}
'''
however the https://github.com/zephyrproject-rtos/zephyr/blob/4793e00ae803ae6b7948e4c110f7ceef11821ef5/cmake/modules/yaml.cmake#L302
will get
"gnuarmemb" it is a string, so line 303 will failure with below error message

'''
PrintARG_YAML_KEYcmake;toolchain;name
CMake Error at D:/internal_zephyr/zephyr/cmake/modules/yaml.cmake:313 (string):
  string sub-command JSON failed parsing json string: * Line 1, Column 1

    Bad escape sequence in string
'''

- **impact**

west build failure on windows for NXP platforms

- **reproduce steps**

```
west build -d build\rt1040 -b mimxrt1040_evk samples\hello_world\
```
see error

- **analysis** 

so if we change line 303 to 
'''
string(LENGTH "${subjson}" index)
'''
then this issue is fixed.

this is a regression issue  exposed by #79118

- additional info

Zephyr version: 27456ed2b57f0114cc976d9d74b98c9762818ca3
CMake version: 3.27.0-rc2

hakehuang, 2024-10-11:
@lucien-nxp  who find this issue

nordicjm, 2024-10-11:
So have you tried with https://github.com/zephyrproject-rtos/zephyr/pull/79647 ?

hakehuang, 2024-10-11:
> So have you tried with #79647 ?

it is not the path issue, it is the  string(JSON index LENGTH "${subjson}")  expect a json string, but we provide a string only.

tejlmand, 2024-10-14:
> it is not the path issue, it is the string(JSON index LENGTH "${subjson}") expect a json string, but we provide a string only.

do try https://github.com/zephyrproject-rtos/zephyr/pull/79647, because the JSON string issue happens exactly because the input to the JSON string is concatenated with a path containing single `\`, which then causes the CMake JSON implementation to fail.

closed 2024-10-14 by hakehuang