zephyrproject-rtos/zephyr #79720: retained_mem and mimxrt

blemouzy, 2024-10-11:
I'm having trouble configuring retained_mem on mimxrt1050_evk with Zephyr v3.7.0.

I follow [documentation example](https://docs.zephyrproject.org/latest/services/retention/blinfo.html) and:

- it fails with external SDRAM

  _mimxrt1050_evk.overlay_
  
    ```c
    / {
        sram@81ff0000 {
            compatible = "zephyr,memory-region", "mmio-sram";
            reg = <0x81ff0000 DT_SIZE_K(64)>;
            zephyr,memory-region = "RetainedMem";
            status = "okay";
    
            retainedmem {
                compatible = "zephyr,retained-ram";
                status = "okay";
                #address-cells = <1>;
                #size-cells = <1>;
    
                boot_info0: boot_info@0 {
                    compatible = "zephyr,retention";
                    status = "okay";
                    reg = <0x0 0x10000>;
                    prefix = [CA FE];
                    checksum = <4>;
                };
            };
        };
    
        chosen {
            zephyr,bootloader-info = &boot_info0;
        };
    };
    
    /*
     Reduce sdram0 usage by 64 kB
    
     Original memory range: 0x80000000 - 0x82000000-1
     New memory range: 0x80000000 - 0x81fff000-1
     */
    &sdram0 {
        reg = <0x80000000 (DT_SIZE_M(32)- DT_SIZE_K(64))>;
    };
    ```

  _build error_
  
    ```
    /home/benjamin/zephyr-rtos/include/zephyr/arch/arm/cortex_m/arm_mpu_mem_cfg.h:72:2: error: #error "Unsupported sram size configuration"
       72 | #error "Unsupported sram size configuration"
    ```

- it also fails with OCRAM

  _mimxrt1050_evk.overlay_
  
    ```c
    / {
        chosen {
            zephyr,bootloader-info = &boot_info0;
        };
    };
    
    /*
     Reduce ocram usage by 1 kB
    
     Original memory range: 0x20200000 - 0x20240000-1
     New memory range: 0x20200000 - 0x2023fc00-1
     */
    &ocram {
        reg = <0x20200000 (DT_SIZE_K(256)- DT_SIZE_K(1))>;
    };
    
    
    &flexram {
        rmem: sram@2023fc00 {
            compatible = "zephyr,memory-region", "mmio-sram";
            reg = <0x2023fc00 DT_SIZE_K(1)>;
            zephyr,memory-region = "RetainedMem";
            status = "okay";
    
            retainedmem {
                compatible = "zephyr,retained-ram";
                status = "okay";
                #address-cells = <1>;
                #size-cells = <1>;
    
                boot_info0: boot_info@0 {
                    compatible = "zephyr,retention";
                    status = "okay";
                    reg = <0x0 0x400>;
                    prefix = [CA FE];
                    checksum = <4>;
                };
            };
        };
    };
    ```

  _build error_
  
    ```
    /home/benjamin/zephyr-rtos/drivers/memc/memc_nxp_flexram.c:66:1: note: in expansion of macro 'BUILD_ASSERT'
       66 | BUILD_ASSERT((OCRAM_TOTAL) == DT_REG_SIZE(OCRAM_DT_NODE), "OCRAM node size is wrong");
    ```

Is there any way to declare retain data section and pass memory size check?

I also don't understand documentation example:

```c
/ {
        sram@2003F000 {
                compatible = "zephyr,memory-region", "mmio-sram";
                reg = <0x2003F000 DT_SIZE_K(1)>;
                zephyr,memory-region = "RetainedMem";
                status = "okay";

                retainedmem {
                        compatible = "zephyr,retained-ram";
                        status = "okay";
                        #address-cells = <1>;
                        #size-cells = <1>;

                        boot_info0: boot_info@0 {
                                compatible = "zephyr,retention";
                                status = "okay";
                                reg = <0x0 0x100>;
                        };
                };
        };

        chosen {
                zephyr,bootloader-info = &boot_info0;
        };
};

/* Reduce SRAM0 usage by 1KB to account for non-init area */
&sram0 {
        reg = <0x20000000 DT_SIZE_K(255)>;
};
```

In this example, `sram0` range is [0x20000000 - 0x2003FC00-1].
So why RetainedMem starts at 0x2003F000?

I also don't understand why `boot_info0` size is only 256 B and not 1 KB (but it is not really an issue).

DerekSnell, 2024-10-14:
Hi @blemouzy ,
> it fails with external SDRAM:
> /home/benjamin/zephyr-rtos/include/zephyr/arch/arm/cortex_m/arm_mpu_mem_cfg.h:72:2: error: #error "Unsupported sram size configuration"
   72 | #error "Unsupported sram size configuration"

That is an MPU error.  ARM's MPU only supports binary sizes (2^x).  The MPU cannot handle a size of `DT_SIZE_M(32)- DT_SIZE_K(64)`.  Instead of reducing the size of `sdram0`, perhaps you can add your `RetainedMem` as a child node within `sdram0` to retain the original size.

> it also fails with OCRAM:
> /home/benjamin/zephyr-rtos/drivers/memc/memc_nxp_flexram.c:66:1: note: in expansion of macro 'BUILD_ASSERT'
   66 | BUILD_ASSERT((OCRAM_TOTAL) == DT_REG_SIZE(OCRAM_DT_NODE), "OCRAM node size is wrong");

A similar issue, the FlexRAM has specific sizes of OCRAM it can support.  See NXP Application Note [AN12077](https://www.nxp.com/docs/en/application-note/AN12077.pdf) for more details.  Your size of 255 KB is not a valid option for OCRAM.

> So why RetainedMem starts at 0x2003F000?

Good question, that looks like an error to me.

blemouzy, 2024-10-15:
Hi @DerekSnell,

> That is an MPU error. ARM's MPU only supports binary sizes (2^x). The MPU cannot handle a size of `DT_SIZE_M(32)- DT_SIZE_K(64)`. Instead of reducing the size of `sdram0`, perhaps you can add your [...]

Understood.

It finally works with the following mimxrt1050_evk overlay:

```
/ {
    chosen {
        zephyr,bootloader-info = &boot_info0;
    };
};

&sdram0 {
    retainedmem {
        compatible = "zephyr,retained-ram";
        status = "okay";
        #address-cells = <1>;
        #size-cells = <1>;

        boot_info0: boot_info@0 {
            compatible = "zephyr,retention";
            status = "okay";
            reg = <0x1fff000 0x1000>;
            prefix = [CA FE];
            checksum = <4>;
        };
    };
};
```

Memory looks correctly filled:

```
x /16x 0x81fff000
0x81fff000:	0x2016feca	0x2000002c	0x01010001	0x00000120
0x81fff010:	0x00012002	0x01200300	0x20040000	0x01020008
0x81fff020:	0x00000000	0x20050000	0x00000004	0x00000040
0x81fff030:	0x00000000	0x00000000	0x00000000	0x00000000
```

and displayed bootloader version is correct.

Note: with Zephyr main (1a89e1640c1464340dfb1d652fdf9c02130b482c), I encountered an issue with flash SPI partitions using mimxrt1050_evk@hyperflash.
It looks like flashing application erased MCUBoot.

> A similar issue, the FlexRAM has specific sizes of OCRAM it can support. See NXP Application Note [AN12077](https://www.nxp.com/docs/en/application-note/AN12077.pdf) for more details. Your size of [...]

Thanks for the AN link.
I tried to create a second OCRAM but it's clear now that is not possible.

> > So why RetainedMem starts at 0x2003F000?
> 
> Good question, that looks like an error to me.

I think it's a bug too.
With same error in `tests/boot/mcuboot_data_sharing/boards/nrf52840dk_nrf52840.overlay`.
I will send a PR.

DerekSnell, 2024-10-16:
Hi @blemouzy ,
> I encountered an issue with flash SPI partitions using mimxrt1050_evk@hyperflash board. It looks like flashing application erased MCUBoot.

Will you be creating a new Issue for this MCUboot topic, and sharing the details to replicate?

Since this Issue about retained_mem seems resolved, can we close it now?  Thanks

closed 2024-10-17 by blemouzy