zephyrproject-rtos/zephyr #79736: JSON library causing Bus Fault with fixed size character array

zzasada, 2024-10-11:
**Describe the bug**
When using a fixed-sized character array inside the json struct it causes a BUS FAULT during the json_obj_encode_buf() function call.

**To Reproduce**
Run the code below.

main.c
```
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
#include <zephyr/data/json.h>

LOG_MODULE_REGISTER(main, LOG_LEVEL_DBG);

/*
// Works as expected
struct accel_config{
        char *mode;
};
*/

// BUS FAULT
struct accel_config{
     char mode[8];
};

const struct json_obj_descr accel_descr[] = {
        JSON_OBJ_DESCR_PRIM(struct accel_config, mode, JSON_TOK_STRING),
};

int main(void)
{
	LOG_INF("JSON Test");

	struct accel_config accel = {
			.mode = "motion"
	};

	char buf[128];

	int rc = json_obj_encode_buf(accel_descr, ARRAY_SIZE(accel_descr), &accel, buf, sizeof(buf));
	if (rc < 0) {
			LOG_ERR("Failed to encode JSON object: %d", rc);
			return rc;
	}

	LOG_INF("Encoded JSON object: %s", buf);

	return 0;
}
```

prj.conf

```
CONFIG_LOG=y
CONFIG_LOG_MODE_DEFERRED=y
# CONFIG_USE_SEGGER_RTT=y
CONFIG_LOG_BUFFER_SIZE=8192

CONFIG_JSON_LIBRARY=y

CONFIG_MAIN_STACK_SIZE=8192
CONFIG_HEAP_MEM_POOL_SIZE=8192

CONFIG_FLASH=y
CONFIG_NVS=y
CONFIG_NVS_LOG_LEVEL_OFF=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y

CONFIG_DEBUG=y
```

**Expected behavior**
json_obj_encode_buf() returns serialized buffer.

**Impact**
annoyance

**Logs and console output**

```
*** Booting nRF Connect SDK v2.7.0-5cb85570ca43 ***
*** Using Zephyr OS v3.6.99-100befc70c74 ***
[00:00:00.252,777] <inf> main: JSON Test
[00:00:00.252,838] <err> os: ***** BUS FAULT *****
[00:00:00.252,838] <err> os:   Precise data bus error
[00:00:00.252,838] <err> os:   BFAR Address: 0x69746f6d
[00:00:00.252,868] <err> os: r0/a1:  0x69746f6d  r1/a2:  0x000089e7  r2/a3:  0x20007e9c
[00:00:00.252,868] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00001021
[00:00:00.252,899] <err> os:  xpsr:  0x61000000
[00:00:00.252,899] <err> os: Faulting instruction address (r15/pc): 0x000089b2
[00:00:00.252,929] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:00.252,960] <err> os: Current thread: 0x200023b0 (unknown)
[00:00:01.521,850] <err> os: Halting system
```

**Environment (please complete the following information):**

 - OS:  Windows
 - Toolchain: nRFConnect for VSCode extension
 Zephyr OS v3.6.99-100befc70c74
 nRF Connect SDK v2.7.0-5cb85570ca43

**Additional context**
building for an nRF52840DK

henrikbrixandersen, 2024-10-15:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

zzasada, 2024-10-16:
I will report it in DevZone.

closed 2024-10-16 by zzasada