zephyrproject-rtos/zephyr #79744: Discussion on Zephyr's Poor Real-Time Performance

lenghonglin, 2024-10-12:
Recently, the well-known embedded systems expert Jacob Beningo wrote the '2024 Real-Time Operating System Performance Report.' The report tested several commonly used RTOSes, including PX5, ThreadX, Zephyr, and FreeRTOS, to evaluate real-time performance through test cases such as coroutine scheduling, preemptive scheduling, memory allocation, message handling, and synchronization processing. Based on the results, Zephyr's real-time performance was nearly the worst. What could be the reason for this?

[image]
[Beningo-RtosPerformance-2024-Report.pdf](https://github.com/user-attachments/files/17348393/Beningo-RtosPerformance-2024-Report.pdf)

cocoeoli, 2024-10-12:
From this result, Zephyr still has a lot of space for performance optimization.

heinwessels, 2024-10-14:
You should read the follow up article by the same person: https://www.beningo.com/is-rtos-performance-affected-by-an-mpu/#

Essentially, in the initial benchmark you posted Zephyr had a Memory Protection Unit enabled (MPU) enabled, which has an performance impact because it adds safety features.

Quote from the article:
> Zephyr jumped from a relative performance against PX5 RTOS of 42% to 70.1%! That’s a 28% improvement in the benchmark and a nearly 67% improvement against itself! All because the MPU and a few additional checks were disabled!
You might argue that disabling the MPU in the benchmark is cheating, but since the other RTOSes in the study did not have them enabled, it really levels the playing field and makes it possible to compare the RTOSes more accurately.