zephyrproject-rtos/zephyr #79772: nrf9131ek/nrf9131 CI failure with Device initialization priority validation failed

nordicjm, 2024-10-14:
**Describe the bug**
Board CI failing, see: https://github.com/zephyrproject-rtos/zephyr/actions/runs/11322012174/job/31482120728?pr=78621

**Expected behavior**
Board to work

**Impact**
Showstopper

@maxd-nordic @aasinclair

maxd-nordic, 2024-10-14:
These tests are failing:
drivers.eeprom.emul.build
drivers.eeprom.build

Now, what makes eeprom tests fail on a board without eeprom?

maxd-nordic, 2024-10-14:
/soc/peripheral@50000000/i2c@a000/npm1300@6b/charger <npm1300_charger_init> is initialized before its dependency /soc/peripheral@50000000/i2c@a000/npm1300@6b <mfd_npm1300_init>

maxd-nordic, 2024-10-14:
charger is sensor priority and the mfd device has its own prio.

maxd-nordic, 2024-10-14:
This patch could fix it:
```
diff --git a/tests/drivers/build_all/eeprom/boards/nrf9131ek_nrf9131.conf b/tests/drivers/build_all/eeprom/boards/nrf9131ek_nrf9131.conf
new file mode 100644
index 00000000000..cbe703cf555
--- /dev/null
+++ b/tests/drivers/build_all/eeprom/boards/nrf9131ek_nrf9131.conf
@@ -0,0 +1 @@
+CONFIG_MFD_INIT_PRIORITY=55
```

maxd-nordic, 2024-10-14:
@nordicjm This fails because this PR https://github.com/zephyrproject-rtos/zephyr/pull/79138 messes with the sensor priority. Maybe you can add the above change to https://github.com/zephyrproject-rtos/zephyr/pull/78621 to fix CI since you are already changing nrf9131ek.

nordicjm, 2024-10-14:
@masz-nordic now a new build error: https://github.com/zephyrproject-rtos/zephyr/actions/runs/11324295360/job/31492031631?pr=78621

maxd-nordic, 2024-10-21:
> @masz-nordic now a new build error: https://github.com/zephyrproject-rtos/zephyr/actions/runs/11324295360/job/31492031631?pr=78621

I recommend this change:
```
diff --git a/drivers/regulator/regulator_npm1300.c b/drivers/regulator/regulator_npm1300.c
index 8adb2d3cb81..aa99cafb725 100644
--- a/drivers/regulator/regulator_npm1300.c
+++ b/drivers/regulator/regulator_npm1300.c
@@ -581,7 +581,7 @@ static int soft_start_set(const struct device *dev, uint8_t soft_start)
 int regulator_npm1300_init(const struct device *dev)
 {
        const struct regulator_npm1300_config *config = dev->config;
-       bool enabled;
+       bool enabled = false;
        int ret = 0;
 
        if (!device_is_ready(config->mfd)) {
```

nordicjm, 2024-10-21:
> I recommend this change:
>
> ```
> diff --git a/drivers/regulator/regulator_npm1300.c b/drivers/regulator/regulator_npm1300.c
> index 8adb2d3cb81..aa99cafb725 100644
> --- [...]

Can you submit a PR with this? It's no longer holding my PRs up, but the board is currently broken

maxd-nordic, 2024-10-21:
https://github.com/zephyrproject-rtos/zephyr/pull/80159

nordic-auko, 2024-10-24:
@nordicjm this warning should be fixed no with https://github.com/zephyrproject-rtos/zephyr/pull/80276

closed 2024-10-24 by nordicjm