zephyrproject-rtos/zephyr #79801: drivers: flash: common: drivers.flash.common.mx25r_high_perf fails

katgiadla, 2024-10-14:
**Describe the bug**
The test `tests/drivers/flash/common/drivers.flash.common.mx25r_high_perf` fails.

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/flash/common/ -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4339-g1ec5ce05f9f8 ***
Running TESTSUITE flash_driver
===================================================================
START - test_flash_copy
Test will run on device mx25r6435f@0
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x20000e38
E: r0/a1:  0x00000001  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000001 r12/ip:  0x00000001 r14/lr:  0x00000001
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200006c0 (test_flash_copy)
E: Halting system
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-4232-g9b5260de9a31`

**Additional context**
Introduced by ccc78912be3ef5c7acdc318ffe39281bcea34cb1

de-nordic, 2024-10-14:
This is again problem with SPI not Flash API nor flash_copy; it seem that sequence: `spi_nor_write_protection_set`->`spi_nor_cmd_write`->`spi_transceive_dt` topples the device over.

de-nordic, 2024-10-14:
The last instruction executed, before this goes into the:
```
fault_s.S:80
0              mrs r0, MSP
```
is nrfx_spim.c:798
```
798         nrfy_spim_xfer_desc_t xfer_desc = *p_xfer_desc; 
```

de-nordic, 2024-10-14:
And there is no data to/from flash operation happening, this is only on attempt to set write protection to false.

de-nordic, 2024-10-14:
OK. The problem seems that there are a lot of buffers allocated on stack and somehow that does not overflow stack set by CONFIG_MAIN_STACK_SIZE but by CONFIG_ZTEST_STACK_SIZE.

Unfortunately there seems to be a real problem with finding such issues, because it is very volatile.
In this case it tends to manifest itself when SPI functions were offloaded to peripheral, because  peripheral would then directly write to stack allocated buffer and seem to destroy it; the problem was that setting CONFIG_MAIN_STACK_SIZE did nothing.

Also the problem would manifest itself differently when the buffer size was decreased, as in such case it could go further into test before failing.

This was hard to find and diagnose. Is there some specific way for troubleshooting such issues?

PR #79814 (2024-10-17): tests/flash/common: Increase ztest stack size
The commit sets CONFIG_ZTEST_STACK_SIZE to 4096.

Fixes #79801
Fixes #78418
closed 2024-10-17 by carlescufi