zephyrproject-rtos/zephyr #79803: Percepio Tracelyzer is not working for zephyr 3.6.0

rahulprajapati-eaton, 2024-10-14:
Recently, I updated Zephyr to version 3.6.0 and started facing issues with Percepio Tracealyzer. Previously, it was working fine with Zephyr 3.2.0. Below are the configurations I added to my application:
```
CONFIG_TRACING=y
CONFIG_PERCEPIO_TRACERECORDER=y
```
When I tried to compile, I encountered the following error:
[image]

Even though I enabled the dependencies, I am still getting a build error.

I want use Percepio Tracealyzer in snapshot mode.

nashif, 2024-10-14:
you should open a bug using the bug template

bdijkstra82, 2024-10-15:
What is the error that you encountered?

dkalowsk, 2024-10-15:
More details will be needed to resolve the issue.  Please use the template as requested already.

rahulprajapati-eaton, 2024-10-29:
I have enable this config
[image]
and got this error
[image]

rahulprajapati-eaton, 2024-10-29:
I clone zephyr and its module again and issue got resolved. It's look like some module was not installed properly. Now everything working fine.

closed 2024-10-29 by rahulprajapati-eaton

bdijkstra82, 2024-10-29:
Probably the Segger module wasn't installed probably. It defines `HAS_SEGGER_RTT`. Note that if you want to use snapshots, you should have
`PERCEPIO_TRC_CFG_STREAM_PORT=PERCEPIO_TRC_CFG_STREAM_PORT_RINGBUFFER`