zephyrproject-rtos/zephyr #79804: Kconfig.defconfig.apollo4p:13: error: couldn't parse 'default $(dt_node_reg_size_int,$(DT_NODE_SRAM),1,K)': macro expanded to blank string

ziqi-zhou02, 2024-10-14:
**Describe the bug**
Followed instructions to install nRF command line tools + nRF connect + VS Code Extensions.
OS: Mac Sonoma 14.5
Errors when building:
`soc/arm/ambiq/apollo4x/Kconfig.defconfig.apollo4p:13: error: couldn't parse 'default $(dt_node_reg_size_int,$(DT_NODE_SRAM),1,K)': macro expanded to blank string`

[image]

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

dkalowsk, 2024-10-15:
Can you reproduce this with a `west build` without using NRF?  Otherwise this likely needs to be routed to the Nordic folks for any solution.

CocoaBob, 2024-10-24:
I suddenly have the same issue, and I didn't change anything in the project...
[image]

CocoaBob, 2024-10-24:
Okay, find the solution from Nordic DevZone https://devzone.nordicsemi.com/f/nordic-q-a/114672/zephyr-compile-error-after-unknown-update, roll back `nRF Kconfig` to **v2024.6.20** solves the problem. I think we could close the ticket for now.

closed 2024-11-01 by nordicjm