zephyrproject-rtos/zephyr #79809: STM32F1 hal lacks some interfaces

Szewcson, 2024-10-14:
**Is your feature request related to a problem? Please describe.**
STM32F1x hal lacks fsmc, i2s, sdio and probably other interfaces.

**Describe the solution you'd like**
Add dtsi definitions like in higher families

**Additional context**
I want to port Zephyr to stm32primer 2 board, I'm currently working on board folder, and thinking about adding drivers, but I think hal is not my cup of tea.

benjaminbjornsson, 2024-10-28:
Hi @Szewcson,

Maybe I can help. Not sure exactly what you're asking for though, can you elaborate?

Szewcson, 2024-10-28:
I want to use fsmc and i2s on stm32f103vet6. I saw that that family don't have that peripherals defined. Also I wonder if driver for fmc handles fsmc.

Szewcson, 2024-10-30:
Which file I should edit to add all the defines? Are there other changes needed to backport those peripherals to lower family?

benjaminbjornsson, 2024-10-30:
I had a look at the I2S and the SPI/I2S registers look almost identical between F1 and F4, so hopefully the driver works with at most minor changes. Also, the I2S pinctrl definitions are missing, and those will have to be generated in the hal_stm32 repository.

benjaminbjornsson, 2024-11-10:
Hi @Szewcson,

I made a draft PR #81181 which currently tries to add I2S support. Unfortunately I don't have F1 hardware, but maybe you can help me with testing?

benjaminbjornsson, 2024-11-10:
I successfully built `samples/drivers/i2s/output` using `waveshare_open103z` with these local changes:

```
diff --git a/boards/waveshare/open103z/waveshare_open103z.dts b/boards/waveshare/open103z/waveshare_open103z.dts
index 72dd744c4c2..41ae01b380f 100644
--- a/boards/waveshare/open103z/waveshare_open103z.dts
+++ b/boards/waveshare/open103z/waveshare_open103z.dts
@@ -82,6 +82,7 @@
 		led3 = &led_4;
 		sw0 = &button;
 		watchdog0 = &iwdg;
+		i2s-tx = &i2s2;
 	};
 };
 
@@ -105,6 +106,14 @@
 	/* usbpre not set: USB clock = 72 / 1.5: 48MHz */
 };
 
+&dma1 {
+	status = "okay";
+};
+
+&dma2 {
+	status = "okay";
+};
+
 &usart1 {
 	pinctrl-0 = <&usart1_tx_pa9 &usart1_rx_pa10>;
 	pinctrl-names = "default";
@@ -130,6 +139,18 @@
 	pinctrl-0 = <&spi2_nss_master_pb12 &spi2_sck_master_pb13
 		     &spi2_miso_master_pb14 &spi2_mosi_master_pb15>;
 	pinctrl-names = "default";
+	status = "disabled";
+};
+
+&i2s2 {
+	pinctrl-0 = <&i2s2_ws_pb12 &i2s2_sd_pb15>;
+	pinctrl-names = "default";
+	status = "okay";
+};
+
+&i2s3 {
+	pinctrl-0 = <&i2s3_ws_pa15 &i2s3_sd_pb5>;
+	pinctrl-names = "default";
 	status = "okay";
 };
```

lilinng-163, 2026-09-11:
I'm using the ALIENTEK Elite STM32F103ZET6 board and need SDIO to access a microSD card. I see there is already a WIP branch (rgallaispou/zephyr, branch issue_79809) that adds the F1 sdmmc1 dtsi node and extends drivers/disk/sdmmc_stm32.c for the F1 series.

Could this work be finalized and merged? It currently depends on the hal_stm32 pinctrl PR and is ~760 commits behind main, so it needs a rebase. I'd be happy to test it on real hardware and report back.