zephyrproject-rtos/zephyr #79833: tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map fails on qemu_x86_tiny

henrikbrixandersen, 2024-10-15:
**Describe the bug**
The `tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map` test suite fails on `qemu_x86_tiny` after commit 121cb49a46eb97ec768860e0e69369d3defc9748.

**To Reproduce**
Steps to reproduce the behavior:
1. `west twister -p qemu_x86_tiny -s tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map`
2. See error below
3. `git revert --no-commit 121cb49a46eb97ec768860e0e69369d3defc9748`
4. `west twister -p qemu_x86_tiny -s tests/kernel/mem_protect/mem_map/kernel.memory_protection.mem_map`
5. See success

**Expected behavior**
Test suite passes.

**Impact**
CI failures on `main` branch.

**Logs and console output**
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS build v3.7.0-4378-g1a89e1640c14 ***
Running TESTSUITE mem_map
===================================================================
START - test_k_mem_map_phys_bare_exec
 SKIP - test_k_mem_map_phys_bare_exec in 0.001 seconds
===================================================================
START - test_k_mem_map_phys_bare_rw
E: Page fault at address 0x403f84d6 (error code 0x3)
E: Access violation: supervisor thread not allowed to write
E: PTE: 0x403f8000 -> 0x0011e000: A 
E: EAX: 0x403f84d6, EBX: 0x0000138b, ECX: 0x0000008a, EDX: 0x0000008a
E: ESI: 0x00000000, EDI: 0x401110c8, EBP: 0x40132fa8, ESP: 0x40132f9c
E: EFLAGS: 0x00000246 CS: 0x0008 CR3: 0x0011b000
E: EIP: 0x40128517
E: call trace:
E:      0: 0x401289c0
E:      1: 0x4010e09d
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x4012c0a0 (test_k_mem_map_phys_bare_rw)
Caught system error -- reason 0
 PASS - test_k_mem_map_phys_bare_rw in 0.026 seconds
===================================================================
START - test_k_mem_map_phys_bare_side_effect
E: Page fault at address 0x403f6000 (error code 0x3)
E: Access violation: supervisor thread not allowed to write
E: PTE: 0x403f6000 -> 0x0011e000: 
E: EAX: 0x403f6000, EBX: 0x40111128, ECX: 0x4011c6b4, EDX: 0x4012c0a0
E: ESI: 0x00000000, EDI: 0x401110c8, EBP: 0x40132fa8, ESP: 0x40132fa4
E: EFLAGS: 0x00000202 CS: 0x0008 CR3: 0x0011b000
E: EIP: 0x40128070
E: call trace:
E:      0: 0x401289c0
E:      1: 0x4010e09d
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x4012c0a0 (test_k_mem_map_phys_bare_side_e)
Caught system error -- reason 0
 PASS - test_k_mem_map_phys_bare_side_effect in 0.004 seconds
===================================================================
START - test_k_mem_map_phys_bare_unmap_reclaim_addr
Mapped (1st time): 0x403f44d6
Mapped (2nd time): 0x403f44d6
 PASS - test_k_mem_map_phys_bare_unmap_reclaim_addr in 0.001 seconds
===================================================================
START - test_k_mem_unmap_phys_bare
E: Page fault at address 0x403f2000 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: EAX: 0x403f2000, EBX: 0x40111150, ECX: 0x00000000, EDX: 0x00002000
E: ESI: 0x00000000, EDI: 0x401110c8, EBP: 0x40132fa8, ESP: 0x40132fa4
E: EFLAGS: 0x00000203 CS: 0x0008 CR3: 0x0011b000
E: EIP: 0x401280dd
E: call trace:
E:      0: 0x401289c0
E:      1: 0x4010e09d
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x4012c0a0 (test_k_mem_unmap_phys_bare)
Caught system error -- reason 0
 PASS - test_k_mem_unmap_phys_bare in 0.004 seconds
===================================================================
TESTSUITE mem_map succeeded
Running TESTSUITE mem_map_api
===================================================================
START - test_k_mem_map_exhaustion
 SKIP - test_k_mem_map_exhaustion in 0.001 seconds
===================================================================
START - test_k_mem_map_guard_after
mapped a page: 0x403f2000 - 0x403f3000
trying to access 0x403f3004
E: Page fault at address 0x403f3004 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: EAX: 0x40115e8a, EBX: 0x403f2000, ECX: 0x00000000, EDX: 0x403f3004
E: ESI: 0x00000000, EDI: 0x401110e4, EBP: 0x40132fa8, ESP: 0x40132fa4
E: EFLAGS: 0x00000216 CS: 0x0008 CR3: 0x0011b000
E: EIP: 0x4012888d
E: call trace:
E:      0: 0x401289c0
E:      1: 0x4010e09d
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x4012c0a0 (test_k_mem_map_guard_after)
Caught system error -- reason 0
 PASS - test_k_mem_map_guard_after in 0.004 seconds
===================================================================
START - test_k_mem_map_guard_before
mapped a page: 0x403ef000 - 0x403f0000
trying to access 0x403eeffc
E: Page fault at address 0x403eeffc (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: EAX: 0x40115e8a, EBX: 0x403ef000, ECX: 0x00000000, EDX: 0x403eeffc
E: ESI: 0x00000000, EDI: 0x401110e4, EBP: 0x40132fa8, ESP: 0x40132fa4
E: EFLAGS: 0x00000216 CS: 0x0008 CR3: 0x0011b000
E: EIP: 0x401287fa
E: call trace:
E:      0: 0x401289c0
E:      1: 0x4010e09d
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x4012c0a0 (test_k_mem_map_guard_before)
Caught system error -- reason 0
 PASS - test_k_mem_map_guard_before in 0.004 seconds
===================================================================
START - test_k_mem_map_unmap

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_protect/mem_map/src/main.c:276: mem_map_api_test_k_mem_map_unmap: (free_mem equal to 0)
no free memory
 FAIL - test_k_mem_map_unmap in 0.002 seconds
===================================================================
START - test_k_mem_map_user
mapped a page: 0x403ec000 - 0x403ed000 (with K_MEM_PERM_USER)
trying to access 0x403ec000 from userspace
mapped a page: 0x403ec000 - 0x403ed000 (without K_MEM_PERM_USER)
trying to access 0x403ec000 from userspace
E: Page fault at address 0x403ec000 (error code 0x7)
E: Access violation: user thread not allowed to write
E: PTE: 0x403ec000 -> 0x0011e000: RW 
E: EAX: 0x403ec000, EBX: 0x4012fffc, ECX: 0x00000000, EDX: 0x00000000
E: ESI: 0x0000003e, EDI: 0x4012ffe4, EBP: 0x4012ffe0, ESP: 0x4012ffc8
E: EFLAGS: 0x00000246 CS: 0x002b CR3: 0x0011b000
E: EIP: 0x40128005
E: call trace:
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x4012c000 (unknown)
Caught system error -- reason 0
 PASS - test_k_mem_map_user in 0.005 seconds
===================================================================
TESTSUITE mem_map_api failed.

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [mem_map]: pass = 4, fail = 0, skip = 1, total = 5 duration = 0.036 seconds
 - SKIP - [mem_map.test_k_mem_map_phys_bare_exec] duration = 0.001 seconds
 - PASS - [mem_map.test_k_mem_map_phys_bare_rw] duration = 0.026 seconds
 - PASS - [mem_map.test_k_mem_map_phys_bare_side_effect] duration = 0.004 seconds
 - PASS - [mem_map.test_k_mem_map_phys_bare_unmap_reclaim_addr] duration = 0.001 seconds
 - PASS - [mem_map.test_k_mem_unmap_phys_bare] duration = 0.004 seconds

SUITE FAIL -  75.00% [mem_map_api]: pass = 3, fail = 1, skip = 1, total = 5 duration = 0.016 seconds
 - SKIP - [mem_map_api.test_k_mem_map_exhaustion] duration = 0.001 seconds
 - PASS - [mem_map_api.test_k_mem_map_guard_after] duration = 0.004 seconds
 - PASS - [mem_map_api.test_k_mem_map_guard_before] duration = 0.004 seconds
 - FAIL - [mem_map_api.test_k_mem_map_unmap] duration = 0.002 seconds
 - PASS - [mem_map_api.test_k_mem_map_user] duration = 0.005 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 102fa38eb51f13728e4ed3acaa3110b6
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK v0.16.8
 - Commit SHA: 1a89e1640c1464340dfb1d652fdf9c02130b482c

**Additional context**
N/A

henrikbrixandersen, 2024-10-15:
Assigned to @nashif as he is the author of commit 121cb49a46eb97ec768860e0e69369d3defc9748.

nashif, 2024-10-15:
there is a fix already https://github.com/zephyrproject-rtos/zephyr/pull/79820

henrikbrixandersen, 2024-10-15:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/79820

closed 2024-10-15 by henrikbrixandersen