zephyrproject-rtos/zephyr #79834: Hostapd AP enable takes 2 seconds more time than before

fengming-ye, 2024-10-15:
https://github.com/zephyrproject-rtos/hostap/commit/e84a82b4efd5de1c297486890c6f96ed18cbd850
is causing 2 seconds more to enable hostapd AP.

hostapd_derive_psk->pbkdf2_sha1 in previous code costs 0.7s but new code costs  more than 2s.

fengming-ye, 2024-10-15:
@krish2718 @VivekUppunda Pls check it by crypto module test in test_sha1 and trace time.
Honestly I have no clue why this happens. So I'm raising bug to track this.

krish2718, 2024-10-15:
> @krish2718 @VivekUppunda Pls check it by crypto module test in test_sha1 and trace time. Honestly I have no clue why this happens. So I'm raising bug to track this.

Thanks. We already have a bug internally as we see the same behaviour with PBKDF2_SHA1 in NCS, sometimes it takes ~13secs, will keep you posted.

henrikbrixandersen, 2024-10-15:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

krish2718, 2024-10-15:
> @krish2718 @VivekUppunda Pls check it by crypto module test in test_sha1 and trace time. Honestly I have no clue why this happens. So I'm raising bug to track this.

I was able to build the tests using:

1.  https://github.com/zephyrproject-rtos/zephyr/pull/78367/ 
2. build command: `west build -p -b nrf7002dk/nrf5340/cpuapp samples/net/wifi/ -- -DCONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_MBEDTLS_PSA=y -DCONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE=y -DCONFIG_NET_TCP_ISN_RFC6528=n`
3. Enable `WIFI_NM_WPA_SUPPLICANT_CRYPTO_TEST` manually. IMHO we should just make this Kconfig as configurable instead of hidden to avoid modifying source files.

But it fails with `supp_psa_crypto_init failed ret -148` i.e., `PSA_ERROR_INSUFFICIENT_ENTROPY`, as there is not entropy PSA driver for nRF boards yet. So, will continue looking at this in NCS and update if I find a fix.

fengming-ye, 2024-10-16:
> 3. Enable `WIFI_NM_WPA_SUPPLICANT_CRYPTO_TEST` manually. IMHO we should just make this Kconfig as configurable instead of hidden to avoid modifying source files.
> 

It is an option that we don't want user to configure. So it will be hidden and only configurable by developer

krish2718, 2024-10-16:
> > 3. Enable `WIFI_NM_WPA_SUPPLICANT_CRYPTO_TEST` manually. IMHO we should just make this Kconfig as configurable instead of hidden to avoid modifying source files.
> 
> It is an option that we don't want user to configure. So it will be hidden and only configurable by developer

IMHO, as it has `TEST` in the name we could make it configurable and if needed add a warning in Cmake, that is what we are doing with othe TEST options, hidden symbols are mainly for internal dependencies and not for hiding configurable options that needs a change.

fengming-ye, 2024-10-17:
> > > 3. Enable `WIFI_NM_WPA_SUPPLICANT_CRYPTO_TEST` manually. IMHO we should just make this Kconfig as configurable instead of hidden to avoid modifying source files.
> > 
> > 
> > It is an option that we don't want user to configure. So it will be hidden and only configurable by developer
> 
> IMHO, as it has `TEST` in the name we could make it configurable and if needed add a warning in Cmake, that is what we are doing with othe TEST options, hidden symbols are mainly for internal dependencies and not for hiding configurable options that needs a change to the files to enable it.

Could you share one example? I can modify it later.

krish2718, 2024-12-08:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Re-looking at this, this issue might be unrelated to the internal bug, without `MBEDTLS_PSA_ACCEL_ALG_PBKDF2_HMAC` HW accelerator MbedTLS uses `PSA_HAVE_SOFT_PBKDF2_HMAC` which is ~50% slower than non-PSA implementation (probably because of added separation due to PSA). So, unless we find a platform that support this accelerator, PSA would be slower than legacy but secure :).

krish2718, 2024-12-13:
Let's close this for now, if we have more data we can re-open.

closed 2024-12-13 by krish2718