zephyrproject-rtos/zephyr #79836: arch: xtensa: Undefined reference to `_xtos_pso_savearea` and `_xtos_core_restore_nw`

ttmut, 2024-10-15:
**Describe the bug**
Building for Xtensa architecture using a recent Cadence toolchain fails with undefined reference errors to `_xtos_pso_savearea` and `_xtos_core_restore_nw` in file `arch/xtensa/core/startup/reset_vector.S`. 

It seems these symbols have been renamed to `xthal_pso_savearea` and `xthal_core_restore_nw`. Build succeeds if above symbols are replaced with these.

**To Reproduce**
Not straightforward to reproduce as it only happens on a custom Xtensa core with `#define XCHAL_HAVE_PSO_CDM 1` and `#define XCHAL_HAVE_PSO_FULL_RETENTION 0`. Sample cores have `XCHAL_HAVE_PSO_CDM` set to 0 so the issue does not occur for them.

**Expected behavior**
Successful build.

**Impact**
Showstopper.

**Logs and console output**
```
xt-ld: zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a(reset_vector.S.obj): in function `__start':
C:/Proj/Zephyr/zephyrproject/zephyr/arch/xtensa/core/startup/reset_vector.S:40: undefined reference to `_xtos_pso_savearea'
xt-ld: C:/Proj/Zephyr/zephyrproject/zephyr/arch/xtensa/core/startup/reset_vector.S:40: undefined reference to `_xtos_core_restore_nw'
xt-ld: zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a(reset_vector.S.obj): in function `_ResetHandler':
C:/Proj/Zephyr/zephyrproject/zephyr/arch/xtensa/core/startup/reset_vector.S:334: undefined reference to `_xtos_core_restore_nw'
clang: error: Xtensa-ld command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

**Environment:**
 - OS: Windows 10
 - Toolchain: Cadence RJ-2024.03
 - Commit SHA: 1a89e1640c1464340dfb1d652fdf9c02130b482c

dcpleung, 2024-10-15:
Should be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/79731

ttmut, 2024-10-23:
The issue still remains. I am using xt-clang compiler.

dcpleung, 2024-10-23:
I don't have access to cores with that combination so I cannot debug this. Could you try removing the whole #if block and see if it still works?

ttmut, 2024-10-23:
It seems to work, though I had to remove 3 #if blocks.

dcpleung, 2024-10-23:
If you need the save/restore feature, you will need to copy the corresponding code from the Xtensa toolchain assembly files into your project (e.g. into your SoC code). Those functions are not provided as part of the HAL but the shutoff/restore assembly files.

ttmut, 2024-10-24:
Thanks for your quick response.

Why not update the function names though, something like below? I am already building Zephyr using Xtensa toolchain so they can be directly linked to prebuilt libhal for the core I am using.
```diff
- movi	a3, _xtos_pso_savearea
+ movi	a3, xthal_pso_savearea
/* Here, as below for _start, call0 is used as an unlimited-range
* jump.
*/
- call0	_xtos_core_restore_nw
+ call0	xthal_core_restore_nw
```

closed 2024-10-29 by mmahadevan108