zephyrproject-rtos/zephyr #79839: tests: watchdog: tests/drivers/watchdog/wdt_basic_api: failure

hakehuang, 2024-10-15:
**Describe the bug**

wdt_basic_test_suite test can not continue to next one.

the problem you're facing:
 - What target platform are you using?
 - frdm_mcxw71
 - What have you tried to diagnose or workaround this issue?
 - NO
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - NA

**To Reproduce**

```
scripts/twister -p  frdm_mcxw71 -T tests/drivers/watchdog
```

**Expected behavior**

all tests PASS

**Impact**

watchdog issue

**Logs and console output**

```
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.0-4339-g1ec5ce05f9f8

decsny, 2024-10-15:
it's good to track this, but I think @hakehuang we found this before internally, the issue was something to do with the lack of ram retention IIRC, we will have to consider a solution for it. It's not really that there is a bug related to the watchdog, though

hakehuang, 2024-10-16:
> if we should track this as a watchdog bug or some kind of enhancement

I think the could be an enhabcement.

PR #94833 (2025-09-16): Retain memory for frdm_mcxw7x tests/drivers/watchdog/wdt_basic_api
Use non-ecc memory to retain test data, fix #79839
In dts, wdog clock source is 1 so the correct clock phandle shall be 32K_CLK.