zephyrproject-rtos/zephyr #79848: In the CDC_ACM sample with the USB device next on the STM32H747i, USB high speed cannot be changed to full speed

summer4821smartbox, 2024-10-15:
**Describe the bug**
I tried to change the USB device speed in the CDC ACM sample from high speed to full speed. The USB device switches to full speed successfully, but the **USB device next** shows an error. And it still be enumerated as high speed.

The issue is caused by the lack of handling for the maximum-speed property from the DTS in the udc_stm32.c .
```
[00:00:02.712,000] <err> usbd_ch9: Configuration descriptor 1 not found
```
**To Reproduce**
1.Add the full-speed setting in the device overlay file.
```
&zephyr_udc0 {
	maximum-speed = "full-speed";
 	cdc_acm_uart0 {
		compatible = "zephyr,cdc-acm-uart";
 	};
};
```
2.Build firmware
```
west build --pristine -b stm32h747i_disco/stm32h747xx/m7 zephyr/samples/subsys/usb/cdc_acm/ -DSHIELD=st_b_lcd40_dsi1_mb1166_a09 -DCONF_FILE=usbd_next_prj.conf
```
3.Observe the error when running the code.

**Expected behavior**
The USB device should be enumerated as a full-speed device and work as expected without errors.

**Environment (please complete the following information):**
Microcontroller: STM32H747I
Zephyr Version: Zephyr main branch Commit 8fda0528
OS: Ubuntu 23.10

marwaiehm-st, 2024-12-09:
Hi @summer4821smartbox 

I tested the issue with the same board and with this overlay "samples/subsys/usb/cdc_acm/stm32h747i_disco_stm32h747xx_m7.overlay" :
```c
&zephyr_udc0 {
	maximum-speed = "full-speed";
 	cdc_acm_uart0 {
		compatible = "zephyr,cdc-acm-uart";
 	};
};
```
I have attempted to reproduce the issue but **have not been able** to see this log error
```shell
[00:00:02.712,000] <err> usbd_ch9: Configuration descriptor 1 not found
``` 
However, I noticed that the **USB_MAXIMUM_SPEED** flag is not defined in the USB device next. When you add the full-speed setting in the device overlay file with '**maximum-speed = "full-speed"**', it will not have any effect because the flag is not handled.

```c
#ifdef USB_MAXIMUM_SPEED
	if (!strncmp(USB_MAXIMUM_SPEED, "high-speed", 10)) {
		speed = USB_OTG_SPEED_HIGH;
	} else if (!strncmp(USB_MAXIMUM_SPEED, "full-speed", 10)) {
#if defined(CONFIG_SOC_SERIES_STM32H7X) || defined(USB_OTG_HS_EMB_PHY)
		speed = USB_OTG_SPEED_HIGH_IN_FULL;
#else
		speed = USB_OTG_SPEED_FULL;
#endif
	} else {
		LOG_DBG("Unsupported maximum speed defined in device tree. "
			"USB controller will default to its maximum HW "
			"capability");
	}
#endif
``` 

To resolve this, try adding the definition for **USB_MAXIMUM_SPEED** in the **udc_stm32.c** file and test with it.

```c
#if DT_INST_NODE_HAS_PROP(0, maximum_speed)
#define USB_MAXIMUM_SPEED      DT_INST_PROP(0, maximum_speed)
#endif
``` 

Could you provide the complete log output from the time the device is connected to when the error occurs? This will help us understand the sequence of events leading to the error.
Are there any other observations or details that you think might be relevant to reproducing the issue?

erwango, 2025-02-03:
@summer4821smartbox Did you had a chance to test again  with proposed fix and provide more logs ?

marwaiehm-st, 2025-04-11:
@summer4821smartbox , Could you please test with this PR https://github.com/zephyrproject-rtos/zephyr/pull/86922 ?

mathieuchopstm, 2025-10-01:
Should be fixed by #96028. Please re-open if issue is still present.

closed 2025-10-01 by mathieuchopstm