zephyrproject-rtos/zephyr #79853: samples: subsys: ipc: ipc_service: icmsg: sample.ipc.icbmsg.nrf54l15_no_multithreading fails

katgiadla, 2024-10-15:
**Describe the bug**
The sample `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icbmsg.nrf54l15_no_multithreading` fails.

Observed for `nrf54l15dk/nrf54l15/cpuapp`

**To Reproduce**
Steps for reproducing:
1. have `nrf54l15dk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/ipc/ipc_service/icmsg -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4339-g1ec5ce05f9f8 ***
I: IPC-service HOST demo started
ASSERTION FAIL [len <= sizeof(rx_buffer)] @ zephyr/subsys/ipc/ipc_service/lib/icmsg.c:190
E: r0/a1:  0x00000004  r1/a2:  0x000000be  r2/a3:  0x200003c8
E: r3/a4:  0x00000004 r12/ip:  0x00000002 r14/lr:  0x00001ab3
E:  xpsr:  0x29000000
E: Faulting instruction address (r15/pc): 0x00006034
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Halting system
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-3952-gb6aed5c505e6`

**Additional context**
Propably introduced by ffe6910e2105e16affd16ee4d2e5967b82eceb53

jaz1-nordic, 2024-10-17:
Fix here:
https://github.com/zephyrproject-rtos/zephyr/pull/79997

katgiadla, 2024-11-04:
The issue occurs still despite of fix:

```
*** Booting Zephyr OS build v4.0.0-rc2 ***
I: IPC-service HOST demo started
ASSERTION FAIL [len <= sizeof(rx_buffer)] @ ../../../../../../../../../zephyr/subsys/ipc/ipc_service/lib/icmsg.c:190
E: r0/a1:  0x00000004  r1/a2:  0x000000be  r2/a3:  0x00000000
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x00001b0f
E:  xpsr:  0x29000000
E: Faulting instruction address (r15/pc): 0x00005600
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Halting system
```

Apart from described issues, the sample `sample.ipc.icbmsg.nrf54l15_remote_no_multithreading` reports timeout from version `v3.7.0-4713-g091c6664c51b`:
```
*** Booting Zephyr OS build v4.0.0-rc2 ***
[00:00:00.000,000] <inf> host: IPC-service HOST demo started`
```
and `sample.ipc.icbmsg.nrf54l15`:
```
*** Booting Zephyr OS build v3.7.0-4790-g6f64f7755796 ***
[00:00:00.436,180] <inf> host: IPC-service HOST demo started
[00:00:00.000,000] <inf> host: Ep bounded
[00:00:00.000,000] <inf> host: Perform sends for 1000 [ms]
```

@masz-nordic, @jaz1-nordic and @doki-nordic could you check it?

katgiadla, 2025-07-03:
Issue still occur for configuration `sample.ipc.icbmsg.nrf54l15[nrf54015dk/nrf54l15/cpuapp]`:
```
"*** Booting Zephyr OS build v4.2.0-rc1-129-gfdd02c5105d2 ***
I: IPC-service HOST demo started
E: Failed to send bound, err -16
```

nordic-piks, 2025-08-20:
@jaz1-nordic Can you help?

closed 2025-11-03 by github-actions[bot]