zephyrproject-rtos/zephyr #79864: tests: subsys: jwt: libraries.encoding.jwt.rsa.psa fails

katgiadla, 2024-10-15:
**Describe the bug**
The test `tests/subsys/jwt/libraries.encoding.jwt.rsa.psa` fails

Observed for:
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/jwt -p nrf9160dk/nrf9160/ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs  --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4339-g1ec5ce05f9f8 ***
Running TESTSUITE jwt_tests
===================================================================
START - test_jwt

    Assertion failed at zephyr/tests/subsys/jwt/src/main.c:48: jwt_tests_test_jwt: (res not equal to 0)
Signing payload
 FAIL - test_jwt in 0.022 seconds
===================================================================
TESTSUITE jwt_tests failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [jwt_tests]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.022 seconds
 - FAIL - [jwt_tests.test_jwt] duration = 0.022 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 8cf83447ecd64014499884f8a73d7731
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-4232-g9b5260de9a31`

**Additional context**
Probably introduced by 29e82d2f23b099ead21a045a869818e24552de87

jukkar, 2024-11-22:
Does not seem to be network related so re-assigning. PSA mentioned, so @tomi-font can you take a look at this?

tomi-font, 2024-11-22:
@valeriosetti touched this, so passing the hot potato to him.

tomi-font, 2024-11-22:
Curious to know, @katgiadla how did you come up with the "probably introduced by" commit?

katgiadla, 2024-11-22:
> Curious to know, @katgiadla how did you come up with the "probably introduced by" commit?

I've used bisection to get _probably_ reason of the bug. It is important to confirm, if the marked commit is real cause of the bug.

valeriosetti, 2024-11-22:
## Short answer
I still haven't found a solution for this, but my suspects are that builds with TF-M does not fully support RSA signature. I need to 

## Explanation

I checked that:
- `west build -p always -b nrf9160dk/nrf9160 --test tests/subsys/jwt/libraries.encoding.jwt.rsa.psa` is OK
- `west build -p always -b nrf9160dk/nrf9160/ns --test tests/subsys/jwt/libraries.encoding.jwt.rsa.psa` FAILS

In both cases I tested on a real device flashing it with `west flash --runner nrfutil` and then looking at the console.
The only difference between the 2 devices is that the 1st one does not have TF-M, while the 2nd one yes.

The same issue happens also if I replace the NRF9160 board with a `mps2/an521/cpu0` one (this is another board that has both the TF-M and non-TF-M images available in Zephyr) and I can test it with QEMU.

valeriosetti, 2024-11-25:
After digging a bit more on this issue, it turned out to be a TF-M limitation: it did not allocate enough heap memory to the crypto partition in order to allow RSA 2048-bit signatures. This should be fixed by #81869

valeriosetti, 2024-11-25:
@katgiadla once #81869 has CI green (a part from the `dnm` label, of course), can you please:
- checkout at current PR
- run `west update`
- re-test the failing test
?

valeriosetti, 2024-11-28:
Reopening because this will be really fixed only when #81869 will be merged as well

closed 2024-12-04 by kartben