zephyrproject-rtos/zephyr #79867: subsys: zbus: publish_stats: message_bus.zbus.publish_stats fails

katgiadla, 2024-10-15:
**Describe the bug**
The test `tests/subsys/zbus/publish_stats/message_bus.zbus.publish_stats` fails

Observed for:
- `nrf52840dk/nrf52840`
- `nrf52dk/nrf52832`
- `nrf54l15dk/nrf54l15/cpuapp`
- `nrf54l15dk/nrf54l15/cpuflpr`
- `nrf5340dk/nrf5340/cpunet`
- `nrf5340dk/nrf5340/cpuapp`
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf9160dk/nrf9160` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/zbus/publish_stats -p nrf9160dk/nrf9160 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs  --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4232-g9b5260de9a31 ***
Running TESTSUITE publish_stats
===================================================================
START - test_channel_metadata

    Assertion failed at zephyr/tests/subsys/zbus/publish_stats/src/main.c:37: publish_stats_test_channel_metadata: 1000 not within zbus_chan_pub_stats_avg_period(&chan) +/- 50

 FAIL - test_channel_metadata in 1.018 seconds
===================================================================
TESTSUITE publish_stats failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [publish_stats]: pass = 0, fail = 1, skip = 0, total = 1 duration = 1.018 seconds
 - FAIL - [publish_stats.test_channel_metadata] duration = 1.018 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: e101656a64ea052518fadcb02f598410
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-4232-g9b5260de9a31`

**Additional context**
Probably introduced by 865f88c8038d062c5facd7c86152de7b81bc5f77

KamilxPaszkiet, 2024-10-17:
reproduction also on:

- it8xxx2_evb
- frdm_k64f 
- intel_adsp/ace15_mtpm
- intel_adsp/ace20_lnl
- up_squared 
- npcx9m6f_evb
- intel_ish_5_4_1
- mec15xxevb_assy6853 
- mec172xevb_assy6906

JordanYates, 2024-10-17:
Can you check if this PR fixes the issue? #79962

katgiadla, 2024-10-17:
> Can you check if this PR fixes the issue? #79962

LGTM and works for nrf platforms

KamilxPaszkiet, 2024-10-17:
> tests/subsys/zbus/publish_stats/message_bus.zbus.publish_stats

I checked this fix on up_squared and it8xxx2_evb.
Test pass

closed 2024-11-06 by dkalowsk