zephyrproject-rtos/zephyr #79871: CMake errors when running tests on Windows via Twister

LukaszMrugala, 2024-10-15:
**Describe the bug**
A clear and concise description of what the bug is.

On Windows, I am unable to use Twister properly. When launching Twister, some tests fail with a CMake error. I have selected the following command to show in logs: `python scripts\twister -T tests\ztest\ -p qemu_x86_64 -ll DEBUG -vv`.

 - What target platform are you using?
   - `qemu_x86_64`, possibly any
 - What have you tried to diagnose or workaround this issue?
   - I have modified the offending `yaml.cmake` file.
   - Printing out values shows that Windows paths cause the problem
   - Adding the following code to the `yaml_set` seems to fix the immediate issue:
     ```
     if(DEFINED ARG_YAML_LIST AND NOT "${ARG_YAML_LIST}" STREQUAL "")
       cmake_path(CONVERT "${ARG_YAML_LIST}" TO_CMAKE_PATH_LIST ARG_YAML_LIST)
     endif()
     ```

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the Zephyr and Twister Getting started Guide on Windows
2. Run `python scripts\twister -T tests\ztest\ -p qemu_x86_64 -ll DEBUG -vv`
3. See two tests failing - `tests/ztest/base/testing.ztest.base.verbose_0_userspace` `tests/ztest/zexpect/testing.ztest.expect_cpp`

**Expected behavior**
Tests do not exhibit opaque CMake errors.

**Impact**
I cannot effectively run any tests on Windows.

**Logs and console output**
twister.log
```
2024-10-10 15:00:57,170 - twister - INFO - Using Ninja..
2024-10-10 15:00:57,865 - twister - INFO - Zephyr version: 27456ed2b57f
2024-10-10 15:00:57,959 - twister - DEBUG - Running cmake script <ZEPHYR_BASE>\cmake\verify-toolchain.cmake
2024-10-10 15:00:57,975 - twister - DEBUG - Calling cmake: ['<CMAKE>\\bin\\cmake.EXE', '-DFORMAT=json', '-P', '<ZEPHYR_BASE>\\cmake\\verify-toolchain.cmake']
2024-10-10 15:00:59,869 - twister - DEBUG - Finished running <ZEPHYR_BASE>\cmake\verify-toolchain.cmake
2024-10-10 15:00:59,869 - twister - INFO - Using 'zephyr' toolchain.
2024-10-10 15:00:59,869 - twister - DEBUG - Reading test case configuration files under <ZEPHYR_BASE>\tests\ztest...
2024-10-10 15:00:59,872 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\base
2024-10-10 15:00:59,884 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\busy_sim
2024-10-10 15:00:59,900 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\error_hook
2024-10-10 15:00:59,900 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\fail
2024-10-10 15:00:59,915 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\summary
2024-10-10 15:00:59,915 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\zexpect
2024-10-10 15:00:59,915 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\ztest_param
2024-10-10 15:00:59,931 - twister - DEBUG - Found possible testsuite in <ZEPHYR_BASE>\tests\ztest\ztress
2024-10-10 15:00:59,931 - twister - DEBUG - No duplicates found.
2024-10-10 15:01:00,247 - twister - DEBUG - Reading platform configuration files under ['<ZEPHYR_BASE>\\/boards', '<ZEPHYR_BASE>\\/subsys/testsuite/boards']...
2024-10-10 15:01:00,390 - twister - DEBUG - platform filter: ['qemu_x86_64']
2024-10-10 15:01:00,390 - twister - DEBUG -   vendor filter: []
2024-10-10 15:01:00,390 - twister - DEBUG -     arch_filter: None
2024-10-10 15:01:00,390 - twister - DEBUG -      tag_filter: None
2024-10-10 15:01:00,390 - twister - DEBUG -     exclude_tag: None
2024-10-10 15:01:00,390 - twister - INFO - Building initial testsuite list...
2024-10-10 15:01:00,468 - twister - INFO - Writing JSON report <ZEPHYR_BASE>\twister-out\testplan.json
2024-10-10 15:01:01,068 - twister - INFO - Adding tasks to the queue...
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/base/testing.ztest.base.verbose_0
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/base/testing.ztest.base.verbose_0_userspace
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/base/testing.ztest.base.verbose_1
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/base/testing.ztest.base.verbose_2
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/error_hook/testing.ztest.error_hook
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/error_hook/testing.ztest.error_hook.no_userspace
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/summary/testing.ztest.summary.shared_unit_test
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/zexpect/testing.ztest.expect
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/zexpect/testing.ztest.expect_cpp
2024-10-10 15:01:01,068 - twister - DEBUG - adding qemu_x86_64\tests/ztest/ztress/testing.ztest.ztress
2024-10-10 15:01:01,068 - twister - INFO - Added initial list of jobs to queue
2024-10-10 15:01:08,496 - twister - DEBUG - Launched 8 jobs
2024-10-10 15:04:33,635 - twister - INFO - 30 test scenarios (27 test instances) selected, 17 configurations skipped (17 by static filter, 0 at runtime).
[... 292 lines cut ...]
  <ZEPHYR_BASE>/cmake/modules/zephyr_default.cmake:133 (include)
  <ZEPHYR_BASE>/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  <ZEPHYR_BASE>/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:10 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at <ZEPHYR_BASE>/cmake/modules/yaml.cmake:307 (foreach):
  foreach Invalid integer: 'NOTFOUND'
Call Stack (most recent call first):
  <ZEPHYR_BASE>/cmake/modules/extensions.cmake:3718 (yaml_set)
  <ZEPHYR_BASE>/cmake/modules/kconfig.cmake:359 (build_info)
  <ZEPHYR_BASE>/cmake/modules/zephyr_default.cmake:133 (include)
  <ZEPHYR_BASE>/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  <ZEPHYR_BASE>/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!

```

**Environment (please complete the following information):**

 - OS:  Windows
 - SDK: Zephyr SDK
 - SHA: 27456ed2b57f0114cc976d9d74b98c9762818ca3

pdgendt, 2024-10-15:
I think this is already fixed in #79647

LukaszMrugala, 2024-10-16:
> I think this is already fixed in #79647

Confirmed on my machine. Closing as fixed.

closed 2024-10-16 by LukaszMrugala