zephyrproject-rtos/zephyr #79897: Wi-Fi DPP code causing build error

fengming-ye, 2024-10-16:
**Describe the bug**
DPP build error when CONFIG_WIFI_NM_WPA_SUPPLICANT_DPP=n and CONFIG_WIFI_NM_HOSTAPD_AP=y

**To Reproduce**
CONFIG_WIFI_NM_WPA_SUPPLICANT_DPP=n and CONFIG_WIFI_NM_HOSTAPD_AP=y
build wifi example

**Expected behavior**
build pass

**Impact**
wifi examples build error

**Logs and console output**
zephyr_downstrem_build/zephyr/modules/hostap/src/supp_api.h:388:56: warning: 'struct wifi_dpp_params' declared inside parameter list will not be visible outside of this definition or declaration
388 | int hapd_dpp_dispatch(const struct device *dev, struct wifi_dpp_params *params);

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 1da74ef705bcf1723a3271266622948251253c0a

**Additional context**

fengming-ye, 2024-10-16:
@krish2718 https://github.com/zephyrproject-rtos/zephyr/pull/79224
is causing build error because hapd_dpp code is not wrapped.
FYI DPP macros are removed as per review requests in https://github.com/zephyrproject-rtos/zephyr/pull/73707.
So why do we need to hide these again? Could you pls share build errors you were trying to fix?

krish2718, 2024-10-16:
> @krish2718 #79224 is causing build error because hapd_dpp code is not wrapped. FYI DPP macros are removed as per review requests in #73707. So why do we need to hide these again? Could you pls share build errors you were trying to fix? In my view zephyr L2 DPP implement is independent from hostap.

I guess by `wrapping` you are referring to using `#ifdef DPP`? The build errors are due to below compilation warning (pasted one, but there are many). and in general if we have a feature flag (optional feature) we need to use that to save memory.

```
include/zephyr/net/wifi_mgmt.h:946:24: error: 'struct wifi_dpp_params::<unnamed union>::wifi_dpp_configurator_add_params' invalid; an anonymous union may only have public non-static data members [-fpermissive]
  946 |                 struct wifi_dpp_configurator_add_params {
```

fengming-ye, 2024-10-16:
> I guess by `wrapping` you are referring to using `#ifdef DPP`? The build errors are due to below compilation warning (pasted one, but there are many). and in general if we have a feature flag [...]

Yes by wrapping I mean "#ifdef DPP".
And these build errors are because union structure defined in structure, which is C++ (or C99?) tools error.
We have a fix toup. I'm raising it.

I agree that we want to save memory, I'll also add macro guard in hostapd DPP.

krish2718, 2024-10-16:
> And these build errors are because union structure defined in structure, which is C++ (or C99?) tools erro

Yes, we have a few samples that use C++ (Matter) they throwed this error.

fengming-ye, 2024-10-16:
> > And these build errors are because union structure defined in structure, which is C++ (or C99?) tools erro
> 
> Yes, we have a few samples that use C++ (Matter) they throwed this error.

Fix is already merged in this https://github.com/zephyrproject-rtos/zephyr/pull/79542

closed 2024-10-16 by fengming-ye