zephyrproject-rtos/zephyr #79935: <err> os: Halting system for use static library

zhang-wenchao, 2024-10-16:
Minimal reproducible example:

pj.conf
```
CONFIG_LOG=y
CONFIG_THREAD_STACK_INFO=y
CONFIG_DEBUG_COREDUMP=y
CONFIG_DEBUG_COREDUMP_BACKEND_LOGGING=y
```

CMakeLists.txt

```
cmake_minimum_required(VERSION 3.20.0)

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})

project(app)

target_sources(app PRIVATE src/main.c)

add_subdirectory(lib)

include_directories(${CMAKE_CURRENT_SOURCE_DIR}/lib)

target_link_libraries(app PRIVATE lib)

```

src/main.c

```
#include <stdio.h>
#include "test.h"

int main(void)
{
  test();
  return 0;
}
```

lib/test.c

```
#include <stdio.h>

void test() {
  printf("test");
}
```

lib/test.h

```
void test();
```

lib/CMakeLists.txt

```
cmake_minimum_required(VERSION 3.20.0)
project(lib)

add_library(lib STATIC
    test.c
)

include_directories(${CMAKE_CURRENT_SOURCE_DIR})

[... 199 lines cut ...]
[00:00:01.885,000] <err> coredump: #CD:666bf58ea1e906361f88d5d525cab157efbd0620e64eb46fc8fd63d8b7619a6e
[00:00:01.895,000] <err> coredump: #CD:12fff7b3ab69993b5bfed72353b0df236e122dbb8ec1c22f7ed2f67a1a8669c2
[00:00:01.905,000] <err> coredump: #CD:d01c343a47531e7e94004a793cc2febed4d42198e3c8b059ff516ffeda6d8f6c
[00:00:01.915,000] <err> coredump: #CD:91a4d9be899755a838196ca06b0cc9cbdf0eefe5f5fb7ecca7b7be08b0bdba0a
[00:00:01.925,000] <err> coredump: #CD:650127722b8e85f301c2d885d4f3778f135c6b01c8dd3258dbc1cc05b934d156
[00:00:01.935,000] <err> coredump: #CD:15231ca2756b7b2a8cbece45224bea85c1e2135202386421f1a2520e2860ebd2
[00:00:01.945,000] <err> coredump: #CD:e155977038697d74a8838df2582d69f7a8d050b1634a0c1584c95aeaf770343f
[00:00:01.955,000] <err> coredump: #CD:17605e77817669f47611ce54f476c1eaf885dd61f2b39c56d536bd4d98ca228b
[00:00:01.965,000] <err> coredump: #CD:1ec67295e2b3c7f37cdc1e1924713e8f9f17e8d297c8a9930b39df35ea66df2f
[00:00:01.975,000] <err> coredump: #CD:c7e9733f1018f6f99267528f16b7cb444528d1bc4d3c0ca57d77683cb76d17a5
[00:00:01.985,000] <err> coredump: #CD:037de623a902785374aef4e61be7e7f2e1b6762014cdf74346692d79ad6be4e7
[00:00:01.996,000] <err> coredump: #CD:42eddfde0400000000000000080000000000000072670010772c000001000000
[00:00:02.006,000] <err> coredump: #CD:00000000772c0000000000006801002000100000000000009c00002068010020
[00:00:02.016,000] <err> coredump: #CD:68010020151800109c00002000000000680100209518001068010020d3500010
[00:00:02.026,000] <err> coredump: #CD:680100200010000008000000111b00104c180020000000000000000000000000
[00:00:02.036,000] <err> coredump: #CD:000000004c180020000000007d35001000000000000000000800000063500010
[00:00:02.046,000] <err> coredump: #CD:0000000000000000726700104c1800207d3500103f4b001004b08de200000000
[00:00:02.056,000] <err> coredump: #CD:380600203806002000d00700a1480010aa01001000000061c05f0010b5350010
[00:00:02.066,000] <err> coredump: #CD:e91876a2080cf8d88424e6836540fce1d9ef5300428e27b77d35001000000000
[00:00:02.076,000] <err> coredump: #CD:00000000c51300107d3500100000000000000000a51300107d350010a8180020
[00:00:02.086,000] <err> coredump: #CD:a81800208d230010d8a6d5bfcd697338b13fc43038060020
[00:00:02.095,000] <err> coredump: #CD:END#
[00:00:02.100,000] <err> os: Halting system

```

zhang-wenchao, 2024-10-16:
This problem is very strange, torture me for a day.

zhang-wenchao, 2024-10-17:
Request investigation, Ask for investigation, I recovered on my nrf52840_mdk, the same problem.

zhang-wenchao, 2024-10-17:
I also have an ESP32C3, there is no problem, normal. I think the problem may appear on ARM compilation or DAPLink or OpenOCD.

zhang-wenchao, 2024-10-17:
Where can this problem be a problem? Request to improve priority.

zhang-wenchao, 2024-10-17:
frdm_k64f I also have a FRDM-K64F, which is also this error, which looks like all ARMs will be like this.

zhang-wenchao, 2024-10-17:
```

Breakpoint 1, main () at /home/life/Develop/test/rp/src/main.c:6
6         test();
(gdb) stepi
test () at /home/life/Develop/test/rp/lib/test.c:2
2       {
(gdb) stepi
0x00000412 in test () at /home/life/Develop/test/rp/lib/test.c:2
2       {
(gdb) stepi
0xfffffe70 in ?? ()
(gdb) stepi
z_arm_usage_fault () at /home/life/Develop/platform/zephyr-project/zephyr/arch/arm/core/cortex_m/fault_s.S:80
80              mrs r0, MSP
(gdb) 
```

nordicjm, 2024-10-17:
Did you try using https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/application_development/external_lib ? Because this is tested and works

zhang-wenchao, 2024-10-17:
This makes me difficult to understand, because it is just a empty test function.

zhang-wenchao, 2024-10-17:
arm can't, risc-v ok.

nordicjm, 2024-10-17:
```
*** Booting Zephyr OS build v3.7.0-4639-g861fdae65dfc ***
Hello World!
mylib says: Hello World!
```
on nrf5340dk (arm cortex m33) so works fine

zhang-wenchao, 2024-10-17:
> arm can't, risc-v ok.

> ```
> *** Booting Zephyr OS build v3.7.0-4639-g861fdae65dfc ***
> Hello World!
> mylib says: Hello World!
> ```
> 
> on nrf5340dk (arm cortex m33) so works fine

thisit's ok, cmake static lib (os: Halting system)

zhang-wenchao, 2024-10-18:
I have positioned some questions,

```
zephyr_get_compile_options_for_lang_as_string(           C options)
```
Use options:
```
add_library(lib STATIC
    test.c
)

target_compile_options(lib PRIVATE -fno-strict-aliasing -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/life/Develop/platform/zephyr-project/sdk/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/life/Develop/platform/zephyr-project/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wformat -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/life/Develop/test/rp=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/life/Develop/platform/zephyr-project/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/life/Develop/platform/zephyr-project=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99)

```

It's ok, The reason is that the compilation option is not applied to add_subdirectory (lib)

zhang-wenchao, 2024-10-18:
@nashif I don't know who can help me. Excuse me.

zhang-wenchao, 2024-10-18:
Only this:

```
cmake_minimum_required(VERSION 3.20.0)
add_library(lib STATIC
    test.c
)

target_include_directories(lib PUBLIC ${CMAKE_CURRENT_SOURCE_DIR})

file(GENERATE OUTPUT "${CMAKE_CURRENT_BINARY_DIR}/compile_options.txt"
     CONTENT "${options}"
)
target_compile_options(lib PRIVATE -mcpu=cortex-m4  )
```

zhang-wenchao, 2024-10-18:
I temporarily solved this problem. 

```
cmake_minimum_required(VERSION 3.20.0)
add_library(lib STATIC
    test.c
)

target_include_directories(lib PUBLIC ${CMAKE_CURRENT_SOURCE_DIR})

message(STATUS "Zephyr Log: ${TOOLCHAIN_C_FLAGS}")
target_compile_options(lib PRIVATE ${TOOLCHAIN_C_FLAGS})
```

Use cmake `add_subdirectory(lib)`  scene is very many, I think it is necessary to create an example.