zephyrproject-rtos/zephyr #79940: json TOK_FLOAT on nordic builds

neutered, 2024-10-16:
**Describe the bug**
 - What target platform are you using?

nordic nrf9515-dk w/ a mod to the aws_iot example from their 2.7.0 release.

 - What have you tried to diagnose or workaround this issue?

the problem is that TOK_OPAQUE and TOK_FLOAT make the same sort of obj descriptor from the JSON_OBJ_DESCR_xxx macros. this has value have the ‘raw’ value of the float (eg, 0x3e340000 for 0.175 or the address of the opaque thingee) which then gets passed to float_ascii_encode() and then append_bytes() as if were a pointer (it’s not). anyway, i have the naive flaot-to-string running on my local branch but not sure if this goes through nordic or zephyr?

**To Reproduce**

neutered, 2024-10-16:
forgot to attach my naive thing that uses the current opaque / float tie.
[json.patch](https://github.com/user-attachments/files/17401444/json.patch)

henrikbrixandersen, 2024-10-22:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

closed 2024-10-22 by henrikbrixandersen

carlescufi, 2024-10-23:
Link to the devzone question: https://devzone.nordicsemi.com/f/nordic-q-a/115751/json-encoding-float

rlubos, 2024-10-23:
@SeppoTakalo Did you encountered any issues with float encoding in LwM2M? I recall json float support was added speicifically for that purpose.

SeppoTakalo, 2024-10-23:
I don't see any problem in JSON encoding.

We have [LightweightM2M-1.1-int-220 - Setting basic information in JSON format](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/net/lib/lwm2m/interop#current-status)

Also, running manually, I can write float value and read back the same value.  Payload seem to be `{"bn":"/3340/0/","e":[{"n":"5521","v":3.14}]}` which looks fine to me.

Json library have been integrated in https://github.com/zephyrproject-rtos/zephyr/pull/46162 so we use that TOK_FLOAT that is mentioned.

I fail to see what is the actual problem. `float_ascii_encode()` is a static function and not part of public API. If there is actual problem in public API then the issue should be reported against that.

Example of how the API is used in LwM2M, follow the call path:
1. https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_rw_json.c#L575-L596
2. https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_rw_json.c#L268-L297
3. https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/utils/json.c#L1168-L1192
4. https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/utils/json.c#L1133-L1158

Maybe this is a API problem.. In `json_obj_encode()` integers and bools are converted but opaque and floats are not.