zephyrproject-rtos/zephyr #79961: fatal error: posix_types.h: No such file or directory in main

dleach02, 2024-10-17:
PR #79884 introduced build failure in main:

```
west build -p -b m2gl025_miv tests/posix/headers -T portability.posix.headers.without_posix_api
```

produces:

```
In file included from C:/work/zephyrproject/zephyr/lib/posix/options/device_io.c:14:
C:/work/zephyrproject/zephyr/include/zephyr/posix/sys/select.h:9:10: fatal error: posix_types.h: No such file or directory
    9 | #include "posix_types.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

dleach02, 2024-10-17:
The test case 

  portability.posix.headers.without_posix_api:
    extra_configs:
      - CONFIG_POSIX_API=n

Disables the CONFIG_POSIX_API which removes the adding of the include path to include/zephyr/posix. The test case sys_select_h.c will directly include zephyr/posix/sys/select.h

What I don't understand is what is the purpose of this test case? You can fix this by using a fully qualified path to posix_types.h inside of select.h but you probably would also want to do that each time posix_types.h is included.

cfriedt, 2024-10-17:
> The test case 
> 
> ```
>   portability.posix.headers.without_posix_api:
>     extra_configs:
>       - CONFIG_POSIX_API=n
> 
> ```
> Disables the CONFIG_POSIX_API which removes the adding of the include path to include/zephyr/posix. The test case sys_select_h.c will directly include zephyr/posix/sys/select.h

I'm not crazy about the prefix to be honest. IMHO, the POSIX headers need to be cleaned up so that the prefix becomes irrelevant and it can be dropped. `CONFIG_POSIX_API` itself should also be deprecated soon.

> What I don't understand is what is the purpose of this test case?

It's mainly to verify that we can include `<sys/select.h>` etc with `CONFIG_POSIX_API` or `<zephyr/posix/sys/select.h>` without.

Not out of a need for this as a "feature" by any means, but to get back some amount of normalcy after the prefix was added.

> You can fix this by using a fully qualified path to posix_types.h inside of select.h but you probably would also want to do that each time posix_types.h is included.

The problem is that we can't have mixed use in the Zephyr repo (just in tests). So we should use the fully-qualified path in other headers.

closed 2024-10-18 by henrikbrixandersen