zephyrproject-rtos/zephyr #79967: Adding an empty mcuboot.overlay file changes the code partition

johannesneyer, 2024-10-17:
**Describe the bug**

Having an empty `mcuboot.overlay` file in the `sysbuild` folder causes the `zephyr,code-partition` in the devicetree to change.

**To Reproduce**

```sh
cd zephyrproject/zephyr/samples/sysbuild/with_mcuboot/
west build -d build1 -b nrf52840dk/nrf52840 --sysbuild
touch ./sysbuild/mcuboot.overlay
west build -d build2 -b nrf52840dk/nrf52840 --sysbuild
```

the output of `diff build{1,2}/mcuboot/zephyr/zephyr.dts` then is:

```
14c14
< 		zephyr,code-partition = &boot_partition;
---
> 		zephyr,code-partition = &slot0_partition;
```

**Expected behavior**

I expect the devicetree to not be affected by an empty mcuboot overlay.

**Impact**

In my case, it produces an unbootable image. 

**Environment (please complete the following information):**

 - OS: Linux 
 - Toolchain: Zephyr SDK 0.16.8
 - Commit: https://github.com/zephyrproject-rtos/zephyr/commit/829c03bcdca378fa2fa55484d16a190d379f754c

johannesneyer, 2024-10-17:
I just realized that this happens because the `mcuboot.overlay` file causes the `zephyrproject/bootloader/mcuboot/boot/zephyr/app.overlay` file to not be loaded anymore. Is this expected behavior?

nordicjm, 2024-10-17:
> I just realized that this happens because the `mcuboot.overlay` file causes the `zephyrproject/bootloader/mcuboot/boot/zephyr/app.overlay` file to not be loaded anymore. Is this expected behavior?

This is expected

closed 2024-10-17 by nordicjm