zephyrproject-rtos/zephyr #80053: tests drivers: can: fails at nrf54h20 - assertion at icmsg

nordic-piks, 2024-10-18:
**Describe the bug**
The tests:
- tests/drivers/can/timing/drivers.can.timing
- tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.txdl_32
- tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance
- tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation
- tests/drivers/can/api/drivers.can.api
- tests/drivers/can/api/drivers.can.api.rtr
- tests/drivers/can/shell/drivers.can.shell
- tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.unused
- tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.txdl_64

Observed for
- nrf54h20dk/nrf54h20/cpuap

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk/nrf54h20/cpuap connected
2. go to your zephyr dir
3. call ./scripts/twister -T  tests/drivers/can/timing -p nrf54h20dk/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
ASSERTION FAIL [len <= sizeof(rx_buffer)] @ ../../../../../../../../../../zephyr/zephyr/subsys/ipc/ipc_service/lib/icmsg.c:190
E: r0/a1:  0x00000004  r1/a2:  0x000000be  r2/a3:  0x00000000
E: r3/a4:  0x00000004 r12/ip:  0x00000001 r14/lr:  0x0e0b8eb3
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x0e0b8ed2
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x2f011148 (icmsg_workq)
E: Halting system
```
There are expected ping/pong messages.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.8
 - Commit SHA or Version used: v3.7.0-4578-g829c03bcdca3

gmarull, 2024-10-18:
Why is icmsg involved in CAN tests?

henrikbrixandersen, 2024-11-02:
Any more information available on this issue?

nordic-piks, 2024-11-07:
Triaged internally

nordic-piks, 2024-11-14:
> Any more information available on this issue?

Unfortunately, currently I cannot provide more info, I will come back.

closed 2024-11-26 by nordic-piks