zephyrproject-rtos/zephyr #80109: drivers/bluetooth/hci/Kconfig: bt-hci-spi: BT_DRV_RX_STACK_SIZE too small.

mikaelj, 2024-10-20:
BT_DRV_RX_STACK_SIZE is too small and can't be configured. It needs to be more than 512 bytes in released versions of Zephyr, or 640 bytes, on the main branch.  Note: I'm using the nRF SDK v2.5.2, which is not the latest, but the driver hasn't changed in any meaningful way since.

zephyr,bt-hci-spi on a nRF9160 to a zephyr,bt-hci-spi-slave on a nRF52810.  bt_spi_rx_thread() in spi.c causes segfault on the host (nRF9160) because of stack overflow. I can nail it down ot the call to bt_spi_transcieve().

Increasing the value in Kconfig to 1024 seems to work.  The rest of my configuration is irrelevant since this is isolated to a specific thread and its task size.

henrikbrixandersen, 2024-10-22:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

jhedberg, 2024-10-22:
@mikaelj are you able to reproduce this with a vanilla upstream tree (not using the nRF SDK). If not, then this is not the right place to report it. If you can reproduce it, would you like to submit a pull request for it? The fix itself (assuming it's necessary upstream) seems simple enough.

mikaelj, 2024-10-23:
@jhedberg @henrikbrixandersen I'll probably not be able to reproduce using a vanilla upstream tree since this is all tightly coupled to the nRF SDK and I don't know what's what.  I'll post it on the dev zone - thanks!

nordic-piks, 2024-11-07:
Triaged internally

closed 2025-05-24 by github-actions[bot]