zephyrproject-rtos/zephyr #80119: tests: twister: twister/twisterlib/package.py : filter error

hakehuang, 2024-10-21:
**Describe the bug**

 File "/home/shared/disk/zephyr_project/zephyr_test/zephyr/scripts/pylib/twister/twisterlib/package.py", line 31, in package
    if t['status'] != TwisterStatus.FILTER:
KeyError: 'status'

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - all platfroms
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?

```
82548410bdc8120f886f864938bdf07e09bbb558 is the first bad commit
commit 82548410bdc8120f886f864938bdf07e09bbb558
Author: Lukasz Mrugala <lukaszx.mrugala@intel.com>
Date:   Wed Sep 11 15:52:56 2024 +0000

    scripts: twister: NOTRUN status

    Adds a new NOTRUN status, which indicates
    that a test was successfully built, but
    not run on account of being not
    runnable in given test instance.

    Signed-off-by: Lukasz Mrugala <lukaszx.mrugala@intel.com>

 scripts/pylib/twister/twisterlib/reports.py      | 11 ++++---
 scripts/pylib/twister/twisterlib/runner.py       | 40 +++++++++++++++++++++---
 scripts/pylib/twister/twisterlib/statuses.py     |  2 ++
 scripts/tests/twister/test_runner.py             | 27 +++++++++-------
 scripts/tests/twister_blackbox/test_coverage.py  |  1 +
 scripts/tests/twister_blackbox/test_platform.py  | 14 ++++++---
 scripts/tests/twister_blackbox/test_printouts.py |  5 ++-
 scripts/tests/twister_blackbox/test_runner.py    | 19 ++++++++---
 8 files changed, 88 insertions(+), 31 deletions(-)

```

**To Reproduce**

```
scripts/twister -j 10 --prep-artifacts-for-testing --package-artifacts frdm_mcxn236_twister_test_only.tar.gz --inline-logs -T samples/hello_world/ -p frdm_mcxn236
```
**Expected behavior**

build package can be generated

**Impact**

board testing

**Logs and console output**

```
{'name': 'samples/hello_world/sample.basic.helloworld', 'arch': 'arm', 'platform': 'frdm_mcxn236', 'path': 'samples/hello_world', 'run_id': 'dcfbef375a81852057a29bb6d328b371', 'runnable': False, 'retries': 0, 'execution_time': '0.00', 'build_time': '6.54', 'testcases': [{'identifier': 'sample.basic.helloworld', 'execution_time': '0.00', 'status': 'not run', 'reason': 'Test was built only'}]}
Traceback (most recent call last):
  File "/home/shared/disk/zephyr_project/zephyr_test/zephyr/scripts/twister", line 215, in <module>
    ret = main(options, default_options)
  File "/home/shared/disk/zephyr_project/zephyr_test/zephyr/scripts/pylib/twister/twisterlib/twister_main.py", line 231, in main
    artifacts.package()
  File "/home/shared/disk/zephyr_project/zephyr_test/zephyr/scripts/pylib/twister/twisterlib/package.py", line 31, in package
    if t['status'] != TwisterStatus.FILTER:
KeyError: 'status'

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux,): 
 - Toolchain (e.g Zephyr SDK, ...): 0.17.0-rc1
 - Commit SHA or Version used: v3.7.0-4797-g98e0e65069c

LukaszMrugala, 2024-10-21:
To fix this, one would need to add `TwisterStatus.NOTRUN` case to this `if`-`elif` chain: https://github.com/zephyrproject-rtos/zephyr/blob/98e0e65069cf62297c4d03e0ad5b543374f5b735/scripts/pylib/twister/twisterlib/reports.py/#L337-L356

hakehuang, 2024-10-22:
@nashif @LukaszMrugala , I raise the priority to high, as this will blocking al board testing with pack method, which is a must have for our board testing.

mmahadevan108, 2024-10-29:
@hakehuang Can you link the PR to this issue if it fixes it.

hakehuang, 2024-10-30:
> @hakehuang Can you link the PR to this issue if it fixes it.

https://github.com/zephyrproject-rtos/zephyr/pull/80221

PR #80221 (2024-10-31): scripts: twister: Fix NOTRUN in test_only
When using the `--build-only` into `--test-only` Twister setup, `NOTRUN` statuses were not properly rerun.

Now they are properly run again if runnable.

Fixes incorrect test statistics in the `--test-only` summary. 
Adds `status` and `reason` fields for all testsuites of the JSON report. 

Fixes #80119
closed 2024-10-31 by mmahadevan108