zephyrproject-rtos/zephyr #80121: mbedtls: build error due to #include location in `modules/crypto/mbedtls/library/x509_crt.c`

thenguyenyf, 2024-10-21:
**Describe the bug**
 A build error occur on Github CI for Renesas RA family after this PR was merged #72804. It seems it is an issue of mbedtls module relevant to location to include the syscall header file, so this preprocessor causes a conflict to the IODefine in HAL external module
[modules/crypto/mbedtls/library/x509_crt.c:1750](https://github.com/zephyrproject-rtos/mbedtls/blob/a78176c6ff0733ba08018cba4447bd3f20de7978/library/x509_crt.c#L1750)
```c
 #define BC              "18"
```
and [drivers/ra/fsp/src/bsp/cmsis/Device/RENESAS/Include/R7FA8M1AH.h](https://github.com/zephyrproject-rtos/hal_renesas/blob/10326518701e25bf336a2eaeb8b5820110e4e6a3/drivers/ra/fsp/src/bsp/cmsis/Device/RENESAS/Include/R7FA8M1AH.h)
```c
union
    {
        __IOM uint8_t ICMR1;           /*!< (@ 0x00000002) I2C Bus Mode Register 1                                    */

        struct
        {
            __IOM uint8_t BC   : 3;    /*!< [2..0] Bit Counter                                                        */
            __OM uint8_t  BCWP : 1;    /*!< [3..3] BC Write Protect(This bit is read as 1.)                           */
            __IOM uint8_t CKS  : 3;    /*!< [6..4] Internal Reference Clock (fIIC) Selection ( fIIC = PCLKB
                                        *   / 2^CKS )                                                                 */
            __IOM uint8_t MTWP : 1;    /*!< [7..7] MST/TRS Write Protect                                              */
        } ICMR1_b;
    };
```
I can build okay without warning when I move these inclusions to the top of the file
[library/x509_crt.c:2688](https://github.com/zephyrproject-rtos/mbedtls/blob/a78176c6ff0733ba08018cba4447bd3f20de7978/library/x509_crt.c#L2688)
```c
#ifdef _WIN32
#ifdef _MSC_VER
#pragma comment(lib, "ws2_32.lib")
#include <winsock2.h>
#include <ws2tcpip.h>
#elif (defined(__MINGW32__) || defined(__MINGW64__)) && _WIN32_WINNT >= 0x0600
#include <winsock2.h>
#include <ws2tcpip.h>
#else
/* inet_pton() is not supported, fallback to software version */
#define MBEDTLS_TEST_SW_INET_PTON
#endif
#elif defined(__sun)
/* Solaris requires -lsocket -lnsl for inet_pton() */
#elif defined(__has_include)
#if __has_include(<sys/socket.h>)
#include <sys/socket.h>
#endif
#if __has_include(<arpa/inet.h>)
#include <arpa/inet.h>
#endif
#endif
```

**To Reproduce**
 Build this sample for any Renesas RA board (such as ek_ra8m1)
```
west build -p -b ek_ra8d1 samples/net/prometheus -T sample.net.prometheus
```

**Expected behavior**

No build warnings

**Impact**
 It makes the Renesas RA boards cannot use the mbedtls with Socket or INET config is enabled. Btw, the Github workflow CI will be return fail for Twister build on this sample: samples/net/prometheus

**Logs and console output**

```
/__w/zephyr/modules/crypto/mbedtls/library/x509_crt.c:1750:25: error: expected identifier or '(' before string constant
 1750 | #define BC              "18"
      |                         ^~~~
/__w/zephyr/modules/hal/renesas/drivers/ra/fsp/src/bsp/cmsis/Device/RENESAS/Include/R7FA8D1BH.h:10594:27: note: in expansion of macro 'BC'
10594 |             __IOM uint8_t BC   : 3;    /*!< [2..0] Bit Counter                                                        */
      |                           ^~
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: ZephyrSDK 0.16.8
 - Commit SHA or Version used: 98e0e65069cf62297c4d03e0ad5b543374f5b735

thenguyenyf, 2024-12-23:
Hi @d3zd3z, @henrikbrixandersen. Can I confirm that any action has been taken for this issue? Should I create a new PR for mbedtls to address it?

henrikbrixandersen, 2024-12-23:
> Hi @d3zd3z, @henrikbrixandersen. Can I confirm that any action has been taken for this issue? Should I create a new PR for mbedtls to address it?

I have little knowledge about mbedtls, but a PR demonstrating your proposed changes is always a good way to move move a solution forward, I think.

closed 2025-03-09 by github-actions[bot]