zephyrproject-rtos/zephyr #80134: [3.7] up_squared: tests/arch/x86/info/arch.x86.info.userspace [timeout]

mdubielx, 2024-10-21:
**Describe the bug**

`9739759fef6a` pass (55,60 s)
`a3266a8b6275` fail (63,30 s)

**Expected behavior**
test should pass

**To Reproduce**
`west twister -p up_squared -s tests/arch/x86/info/arch.x86.info.userspace`

**Logs and console output**
```
Version 2.18.1263. Copyright (C) 2018 American Megatrends, Inc.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                Press <DEL> or <ESC> to enter setup.
>>Checking Media Presence......
>>Media Present......
>>Start PXE over IPv4. Press ESC key to abort PXE boot.
taIP address is 192.168.24.1
Server IP address is 192.168.21.11
NBP filename is up_squared.efi
NBP filesize is 2448705 Bytes
>>Checking Media Presence......
>>Media Present......
Downloading NBP file...
Succeed to download NBP file.
*** Zephyr EFI Loader ***
RSDP found at 0x79c2e000
Zeroing 11685888 bytes of memory at 0x12a000
Zeroing 3392 bytes of memory at 0xe752c0
Copying 16384 data bytes to 0x1000 from image offset 
Copying 172032 data bytes to 0x100000 from image offset 16384
Copying 2253504 data bytes to 0xc4f000 from image offset 188416
Jumping to Entry Point: 0x116b (48 31 c0 48 31 d2 48)
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build a3266a8b6275 (delayed boot 500ms) ***
info: the Zephyr x86 platform information tool
MULTIBOOT: info struct UNAVAILABLE or EMPTY.
No basic memory map available.
No extended memory map available.
Framebuffer data not present.
MEMORY MAP: source is default, max 64 entries
0000000000000000 -> 000000007fffffff RAM (2097152K)
Kernel exclusions:
0x1000 -> 0x2000 (locore)
0x125000 -> 0xe76000 (ram)
0x125000 -> 0x12a000 (app_smem)
ACPI: 4 CPUs found
CPU #0: APIC ID 0x00
CPU #1: APIC ID 0x02
CPU #2: APIC ID 0x04
CPU #3: APIC ID 0x06
Intel VT-D Supported:
-> X2APIC does not need to be opted out
-> Interrupt remapping supported
[ Hardware Unit Definition 0 ]
- Includes only listed PCI devices under given Segment
- Segment number 0
- Base Address 0xfed64000
- Device Scopes:
. Scope type PCI Endpoint
. Enumeration ID 0
. PCI Path 00:02.00
[ Hardware Unit Definition 1 ]
- Includes all PCI devices under given Segment
- Segment number 0
- Base Address 0xfed65000
- Device Scopes:
. Scope type IOAPIC
. Enumeration ID 1
. PCI Path fa:1f.00
. Scope type MSI Capable HPET
. Enumeration ID 0
. PCI Path 00:1f.07
TIMER: HPET, configured frequency = 19200000Hz
Using CMOS RTC as reference clock:
start = 586431184, end = 605631719, 19200535 cycles
start = 624833231, end = 644035092, 19201861 cycles
start = 663236057, end = 682435787, 19199730 cycles
start = 701637696, end = 720839438, 19201742 cycles
start = 740039906, end = 759240946, 19201040 cycles
start = 778441427, end = 797642434, 19201007 cycles
start = 816843529, end = 836045081, 19201552 cycles
start = 855245516, end = 874446405, 19200889 cycles
start = 893648215, end = 912848456, 19200241 cycles
```

**Environment:**
 
- Linux Ubuntu 22.04
- Toolchain: Zephyr SDK 16.8
- Zephyr version: a3266a8b6275

henrikbrixandersen, 2024-10-22:
Can you reproduce this issue on the `main` branch?

arikgreen, 2024-10-23:
No reproduce on the main branch. 
```
Tue, 22 Oct 2024 03:10:29 GMT    INFO    - 1831/2056 up_squared    tests/arch/x86/info/arch.x86.info.userspace    PASSED (device: up2, 63.517s)
```
and it has sporadic reproduce on the v3.7-branch. 
IMO the test has the execution time close to the time out (60s) and sometimes will it not be able to be do.

Zephyr version: a3266a8b6275:
```
INFO    - 2038/2039 up_squared    arch/x86/info/arch.x86.info    	   PASSED (device: up2, 54.550s)
INFO    - 2039/2039 up_squared    arch/x86/info/arch.x86.info.userspace    PASSED (device: up2_2, 55.367s)
```

laurenmurphyx64, 2024-10-24:
Assigning to @arikgreen as he contributed PR

closed 2024-10-30 by mmahadevan108