zephyrproject-rtos/zephyr #80167: lib: net_buf: Generates lots of warnings during Bluetooth initialization

HoZHel, 2024-10-21:
**Describe the bug**

Several warnings are generated on the console during Bluetooth initialization with the message as follow:
 `<wrn> net_buf: Timeout discarded. No blocking in syswq`.

I discovered this using the platform `nrf52840dk/nrf52840` with [peripheral_hr](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/peripheral_hr), even if the problem is not specific to this application.

Apparently this has been introduced by https://github.com/zephyrproject-rtos/zephyr/commit/88ee2bee393666ce661b47dae3b65bebdfda32d4

**To Reproduce**

1. `west build -b nrf52840dk/nrf52840 samples/bluetooth/peripheral_hr`
2. `west flash`

**Expected behavior**
Normally, I would expect no warning for a regular Bluetooth initialization.

**Impact**
Annoyance and feeling that something is not right.

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-4797-g98e0e65069cf ***
[00:00:00.250,518] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.250,640] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.250,762] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.250,915] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,068] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,190] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,312] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,464] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,586] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,708] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,861] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.251,983] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.252,105] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.252,136] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.252,136] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.7 Build 99
[00:00:00.252,197] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.252,349] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.252,471] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.252,624] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.252,746] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.252,929] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.253,112] <inf> bt_hci_core: Identity: F4:EC:64:8F:85:C8 (random)
[00:00:00.253,143] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.253,173] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
Starting Legacy Advertising (connectable and scannable)
[00:00:00.253,295] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.253,448] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.253,601] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.253,845] <wrn> net_buf: Timeout discarded. No blocking in syswq
Advertising successfully started
Checking LED device...done.
Configuring GPIO pin...done.
Start blinking LED...

```
**Environment:**

 - OS: Windows
 - Toolchain: Zephyr SDK
 - Commit SHA: 98e0e65069cf62297c4d03e0ad5b543374f5b735
 - Version used: 3.7.99

henrikbrixandersen, 2024-10-21:
CC: @JordanYates

JordanYates, 2024-10-21:
More accurately this was introduced by #71697, and any such usage should have been cleaned up as part of that PR.

msmttchr, 2024-10-22:
@JordanYates, I think the change of the message level from debug to warning, triggered all these messages on the console.

JordanYates, 2024-10-22:
> @JordanYates, I think the change of the message level from debug to warning, triggered all these messages on the console.

Yes, but it's the usage of blocking APIs on the system workqueue that is the problem, not the log level of the warning.

krish2718, 2024-12-17:
If the intention is to just let the application developer know that there is a problem in the usage of context, then can't we do Linux style `printk_once`? or `WARN_ONCE` , so, `LOG_WRN_ONCE`, this is a nice tradeoff b/w intimation of problem but not overwhelming console with logs? Of course, there is a caveat that this would only point out a single problematic caller, so, maybe we can log once per-thread?

PavelVPV, 2025-02-20:
I think this issue should be closed as the problematic change is now reverted: https://github.com/zephyrproject-rtos/zephyr/pull/84282

closed 2025-02-20 by jhedberg