zephyrproject-rtos/zephyr #80183: mbedtls:psa  Use psa_crypto_init() asks for entropy even it is not needed for every use case

zhang-wenchao, 2024-10-22:
```
OS: Debian 12
Toolchain:  Zephyr SDK 0.16.8
Commit SHA or Version used: v3.7.0-4339-g1ec5ce05f9f8 v3.7.0-4638-g9d0486e3ee86
```

ref:

https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/HD2QPWESQWLBXYCCU5TXBFZF2ROEA6S5/

https://github.com/Mbed-TLS/mbedtls/issues/6006

https://github.com/Mbed-TLS/mbedtls/issues/6007

zhang-wenchao, 2024-10-22:
```
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_PSA_CRYPTO_C=y
CONFIG_MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG=y
CONFIG_MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG_ALLOW_NON_CSPRNG=y
```

CONFIG_MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG_ALLOW_NON_CSPRNG It doesn't work.

zhang-wenchao, 2024-10-22:
@valeriosetti I'm very sorry to bother you as I see you've had related work before, is this a known issue please?

valeriosetti, 2024-10-22:
> @valeriosetti I'm very sorry to bother you as I see you've had related work before, is this a known issue please?

No worries! Yes, `psa_crypto_init()` always requires some entropy source in order to initialize properly even if you won't need any randomness in your code. That's a limitation and indeed both issues mentioned on the mailing list discussion that you added in the description are not solved yet (they don't actually have a tentative solving PR neither).

However I'm a bit surprised that `CONFIG_MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG_ALLOW_NON_CSPRNG` does not work as expected. It surely not the best choice for true random number generators, but it should work for basically all platforms for a basic build-test. Can you tell me which example is failing so that I can investigate this issue locally and provide some advice?

closed 2024-10-22 by zhang-wenchao