zephyrproject-rtos/zephyr #80193: cmake: KERNEL_VERSION_CUSTOMIZATION and APP_VERSION_CUSTOMIZATION do not work anymore

mlasch, 2024-10-22:
https://github.com/zephyrproject-rtos/zephyr/pull/62395 introduced a feature to set additional defines in `version.h` and `app_version.h`. This seems to be broken, at least in the current version (beginning with Zephyr 3.6.0 AFAIK).

**Steps to reproduce**
Build with `KERNEL_VERSION_CUSTOMIZATION`
```
west build -palways -b nrf52840dk/nrf52840 zephyr/samples/hello_world -- -DKERNEL_VERSION_CUSTOMIZATION="#define;BANNER_VERSION;STRINGIFY(blah)"
```
as mentioned in the PR. Then find a custom define in `build/zephyr/include/generated/version.h`. Note: in Zephyr 3.7.0 `version.h` moved to a new location: `build/zephyr/include/generated/zephyr/version.h`.

The regression appears to be present already in Zephyr 3.6.0.
```
...
#define KERNEL_VERSION_STRING  "3.5.0"

#define BUILD_VERSION zephyr-v3.5.0
#define BANNER_VERSION STRINGIFY(blah)
...
```
```
...
#define KERNEL_VERSION_STRING           "3.6.0"
#define KERNEL_VERSION_EXTENDED_STRING  "3.6.0+0"
#define KERNEL_VERSION_TWEAK_STRING     "3.6.0+0"

#define BUILD_VERSION v3.6.0
<missing>
...
```

Also there seems to be no documentation about this feature.

**Expected behavior**
`KERNEL_VERSION_CUSTOMIZATION` should produce an entry in the `version.h` header file. Same for `APP_VERSION_CUSTOMIZATION` in app_version.h.

**Impact**
`KERNEL_VERSION_CUSTOMIZATION` and `APP_VERSION_CUSTOMIZATION` cannot be used.

rettichschnidi, 2024-10-22:
Seems like this feature can no only be used via CMake properties, not via CLI anymore.

Example:

```console
$ git diff samples/hello_world/CMakeLists.txt
diff --git a/samples/hello_world/CMakeLists.txt b/samples/hello_world/CMakeLists.txt
index ecb7d24bb8f..9b22926d66e 100644
--- a/samples/hello_world/CMakeLists.txt
+++ b/samples/hello_world/CMakeLists.txt
@@ -6,3 +6,4 @@ find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
 project(hello_world)

 target_sources(app PRIVATE src/main.c)
+set_target_properties(version_h PROPERTIES KERNEL_VERSION_CUSTOMIZATION "#define;BANNER_VERSION;STRINGIFY(blah)")
```

Resulting Build:
```console
$ west build --pristine -b nrf52840dk/nrf52840 zephyr/samples/hello_world
<snip>
$ rg BANNER_VERSION build/zephyr/include/generated/zephyr/version.h
23:#define BANNER_VERSION STRINGIFY(blah)
```

tejlmand, 2024-11-01:
@mlasch fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/80743

PR #80743 (2024-11-05): cmake: use CMake variable KERNEL_VERSION_CUSTOMIZATION for default value
Fixes: #80193

With #62395 the Zephyr kernel and app version customization values were moved to target properties to allow Zephyr modules to adjust the values.

This had the consequence described by #80193 that version customization using CMake arguments, `-D`, or CMakeList.txt toplevel file can no longer be used for customizing the version.

To support both CMake variable as well as Zephyr module version customization use-cases then this commit uses `zephyr_get()` to fetch any CMake variable adjustments and uses the valu [... cut]
commit f2ea8506 (2024-11-05): cmake: use CMake variable KERNEL_VERSION_CUSTOMIZATION for default value

Fixes: #80193

With #62395 the Zephyr kernel and app version customization values were
moved to target properties to allow Zephyr modules to adjust the values.

This had the consequence described by #80193 that version customization
using CMake arguments, `-D`, or CMakeList.txt toplevel file can no
longer be used for customizing the version.

To support both CMake variable as well as Zephyr module version
customization use-cases then this commit uses `zephyr_get()` to fetch
any CMake variable adjustments and uses the val [... cut]
closed 2024-11-05 by mmahadevan108