zephyrproject-rtos/zephyr #80203: stm32h5: CONFIG_NET_CONFIG_SETTINGS in combination with a watchdog keeps rebooting the device

javlands, 2024-10-22:
**Describe the bug**
I encountered  a bug where the **stm32h573i_dk** keeps rebooting. I minimized the problem to the combination of `CONFIG_NET_CONFIG_SETTINGS` and a watchdog. Disabling `CONFIG_NET_CONFIG_AUTO_INIT` fixes the issue, but is not desired in our case.

**To Reproduce**
I'm able to reproduce it with the watchdog sample, where I add the config
1. `cd zephyr/samples/drivers/watchdog`
2. `west build -p -b stm32h573i_dk -- -DCONFIG_NETWORKING=y -DCONFIG_NET_CONFIG_SETTINGS=y`
3. `west flash`

**Expected behavior**
The device reboots every +-100ms. The console looks like this:
```
*** Booting Zephyr OS build v3.7.0-97-g9ffdddec37a6 ***
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131

*** Booting Zephyr OS build v3.7.0-97-g9ffdddec37a6 ***
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131

*** Booting Zephyr OS build v3.7.0-97-g9ffdddec37a6 ***
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131

*** Booting Zephyr OS build v3.7.0-97-g9ffdddec37a6 ***
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131

...
```

**Impact**
We are not able to use the auto initialisation for networking.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: zephyr sdk 0.16.8
 - checked out @ 6773f3344575286c8cc268ea15ebac2ded4f9848

FRASTM, 2024-10-23:
Can you try with `CONFIG_WDT_DISABLE_AT_BOOT=y`
else the wdg is starting before the phy is detected "update_link_state" etc

javlands, 2024-10-23:
> Can you try with `CONFIG_WDT_DISABLE_AT_BOOT=y` else the wdg is starting before the phy is detected "update_link_state" etc

This seems to fix the issue, thank you!

javlands, 2024-10-23:
Is this something that should be added to the boards `_defconfig` files?

FRASTM, 2024-10-23:
> Is this something that should be added to the `_defconfig` file of the boards?

not necessarily, CONFIG_WDT_DISABLE_AT_BOOT=y is well set in the prj.conf

erwango, 2024-10-24:
> We were wondering if there is another solution than disabling wdt at boot, as this feels a bit like hiding the problem instead of solving it?

Would you have any other alternative in mind ?

erwango, 2024-10-24:
> IWDG_STM32_INITIAL_TIMEOUT could be set to a higher value when NETWORKING is enabled

This is a good alternative, but this is something that can be done at application/out of tree level.
In upstream, we cannot provide settings that will match all possible use cases combinations.

FRASTM, 2024-11-04:
> solution

When the wdg is enabled at boot, it means during the wdg driver init phase and must _immediately_ be fed by the application. However with networking enabled, the init phase of the networking driver takes a long time (especially with no RJ45 plugged) before going to run the application that will feed the WDG.

FRASTM, 2024-11-04:
> We were wondering if there is another solution than disabling wdt at boot, as this feels a bit like hiding the problem instead of solving it?

When the wdg is enabled at boot, it means during the wdg driver init phase and must immediately be fed by the application. However with networking enabled, the init phase of the networking driver takes a long time (especially with no RJ45 plugged) before going to run the application that will feed the WDG.

erwango, 2024-11-04:
@javlands I'm closing the issue, based on previous explanations.

closed 2024-11-04 by erwango