zephyrproject-rtos/zephyr #80219: Nominate doki-nordic as collaborator

aescolar, 2024-10-22:
# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Dominik Kilian
1. [doki-nordic](https://github.com/doki-nordic) 
1. Nordic Semi

## Supporting Documents

* https://github.com/zephyrproject-rtos/zephyr/pull/58741
* https://github.com/zephyrproject-rtos/zephyr/pull/59371
* https://github.com/zephyrproject-rtos/zephyr/pull/44632
* https://github.com/zephyrproject-rtos/zephyr/pull/71492
* https://github.com/zephyrproject-rtos/zephyr/pull/72377
* https://github.com/zephyrproject-rtos/zephyr/pull/37361

stephanosio, 2024-10-22:
@doki-nordic has been added to the GitHub `collaborators` team.

closed 2024-10-22 by stephanosio

aescolar, 2024-10-22:
@stephanosio, @doki-nordic 's reviews still appeare grayed out, and he is not a member in https://github.com/orgs/zephyrproject-rtos/people?query=doki-nordic
should he be there?

doki-nordic, 2024-10-23:
I accepted yesterday. I think, it was just after you checked.