zephyrproject-rtos/zephyr #80233: Nominate rugeGerritsen as collaborator

aescolar, 2024-10-22:
# Nomination

## GitHub User

1. Rubin Gerritsen
1. [rugeGerritsen](https://github.com/rugeGerritsen)
1. Nordic Semi

## Supporting Documents

* https://github.com/zephyrproject-rtos/zephyr/pull/78897
* https://github.com/zephyrproject-rtos/zephyr/pull/78284
* https://github.com/zephyrproject-rtos/zephyr/pull/78267
* https://github.com/zephyrproject-rtos/zephyr/pull/77925
* https://github.com/zephyrproject-rtos/zephyr/pull/78675
* https://github.com/zephyrproject-rtos/zephyr/pull/78092
* https://github.com/zephyrproject-rtos/zephyr/pull/77704

stephanosio, 2024-10-22:
@rugeGerritsen is not currently listed as a collaborator in the MAINTAINERS file.

aescolar, 2024-10-22:
> @rugeGerritsen is not currently listed as a collaborator in the MAINTAINERS file.

Correct. I will add him (but this is a bit of a chicken and egg :) )

aescolar, 2024-10-24:
@stephanosio the maintainers file now lists @rugeGerritsen :)

stephanosio, 2024-10-25:
@rugeGerritsen has been added as a Maintainer.

closed 2024-10-25 by stephanosio